RobotWebTools/rosbridge_suite #565: websocket.launch doesn't work...

arizk2605, 2021-04-07:
Hi everyone, 
I'm currently working on a streaming server that publishes health data from a wristband (Empatica E4) to a TCP client (More information on it : http://developer.empatica.com/windows-streaming-server.html
My goal is to publish those real-time data messages in a ROS topic so that i can save them into rosbags afterwards.

I tried using the rosbridge_server package included in  rosbridge_suite but the launch files didn't work because of some random errors related to Msg file that i can't seem to find a solution for .

Do you have any idea how i could achieve that ?

abhishek47kashyap, 2021-04-15:
> launch files didn't work because of some random errors related to Msg file

Can you post the errors?

arizk2605, 2021-04-15:
Hi abishek,
Here is a screenshot of the launch file (I updated the host and port accordingly to the server address and port of the E4 streaming server which are respectively **127.0.0.1** and **28000**) I am running this TCP server from a windows computer whereas the launch file is executed from a raspberry pi that I command by ssh using putty. Both are connected to the same network which is my house's.
[image]

And I get this error when launching...
[image]

My goal would be to the publish the data published to the TCP clients from the Empatica E4 server to a ROS topic so that I can save it into rosbags afterwards

abhishek47kashyap, 2021-04-15:
Hmm `No module name rosauth.srv` kinda rings a bell, I think I had to install _rospy-tutorials_ and _roscpp-tutorials_:
`sudo apt-get install -y ros-$ROS_DISTRO-rospy-tutorials ros-$ROS_DISTRO-roscpp-tutorials`

arizk2605, 2021-04-15:
I tried installing it but it failed : 
[image]

Also, I think the main problem here is the "ImportError: No module named msg"

arizk2605, 2021-04-19:
Got any clue ?

abhishek47kashyap, 2021-04-20:
Just noticed, the title refers to _websocket.launch_ but screenshot of the launch file you provided is that of _rosbridge_tcp.launch_ .. out of the [3 launch files](https://github.com/RobotWebTools/rosbridge_suite/tree/develop/rosbridge_server/launch) which one do you want?

I haven't used `rosbridge_tcp` but after a [proper installation of rosbridge](http://wiki.ros.org/rosbridge_suite/Tutorials/RunningRosbridge#Installing_Rosbridge), `roslaunch rosbridge_server rosbridge_websocket.launch` should work!

arizk2605, 2021-04-20:
I tested two of them and I got the same errors for both..
Knowing that the streaming server publishes data to TCP clients, I think rosbridge_tcp is the most suitable.
I followed every step of the tutorial though..

arizk2605, 2021-04-20:
I created another issue, if you can check it please : https://github.com/RobotWebTools/rosbridge_suite/issues/566

closed 2022-03-24 by github-actions[bot]