RobotWebTools/rosbridge_suite #569: Fine-granular access control

uahic, 2021-05-12:
Rosauth works but is a binary access control which either gives you access to all or none 'ressources' accessible via the bridge.
I know that ROS 1 was not really intended to be used in production systems but we do have an industrial project that relies on ROS and the client requested different levels of access control. 
Currently I see two ways
1) creating multiple bridges, each with different filters on topics/services/... (let's call them 'ressources') and request authorization from a webserver, which replies with a token that is accepted by the bridge, corresponding to a particular access-level (e.g. 'admin')

2) relay restricted ressources through a custom webserver (thereby actually creating a second kind of ros-bridge for the poor man)

None of these alternative solutions is really 'clean' 
An implementation in the bridge would probably require appending the client some sort of authentication (token,... like JWT?), and needs to be checked on each request. It does not to query each time another backend server. The bridge just needs to know:  1) the mapping level <-> ressources  2) authenticate the user

Finally, I would like to ask you for your opinions. Possibly I could do an initial implementation if people disapprove with this feature

## Specifications

  - ROS Version noetic
  - OS Version Ubuntu 20.04
  - Rosbridge Version - most recent

reinzor, 2021-05-17:
Hi @uahic , we encountered the same problem you wrote up here. We chose the first solution having a `client` bridge and an `admin` bridge and separate the allowed data streams via the configured `globs`.

I agree that it would be interesting to have a more generic set-up for authentication however this would have quite some impact on how things are handled in the bridge side but definitely worth some thought. I think we could still use the original [rosauth](https://github.com/GT-RAIL/rosauth/blob/develop/srv/Authentication.srv) service interface with a different implementation, e.g. JWT authorization. It already includes some `level` setup.

zflat, 2021-06-09:
A relatively simple approach would be include changes to rosauth and rosbridge.

# rosauth

* Allow configuring with multiple secret files based on user "level"
* When an authentication request is made, read the user "level" from the request to determine the secret to use

# rosbridge

* When the client is authenticated, store the user "level" in addition to just True/False authenticated
* Modify the `topics_glob` (and other similar globs) so that they are scoped to specified user "levels"
* Modify the `topics_glob` to have more granularity for read vs. write. So give the option to specify `topics_read_glob`, `topics_write_glob` as separate lists
  * Use the `topics_read_glob` when subscribing
  * Use the `topics_write_glob` when advertising
* Update checking the glob to be able to take into account the current user level that has been authenticated

iwanders, 2021-06-25:
Hi, I've just filed a PR that may be of interest; #572. It provides a way to specify for each topic glob whether it is read only, write only or whether it's a bidirectional topic. It doesn't provide any client authentication systems as are proposed here, but it does allow you to prevent clients from publishing to topics while still being able to subscribe to others. It's backwards compatible with current launch files. And it does work in a similar way as @zflat suggested with different globs for read and write.