RobotWebTools/rosbridge_suite #576: How to play audio_common message with web audio api?

TobiMiller, 2021-07-28:
Hey,

rosbridge is working like expected. I can subscribe to images and display them like in your tutorials.
i want to send audio from a pc running rosbridge to a web page by using the audio_capture package. i get a wave stream which results as an base64 encoded stream on javascript side.

has anyone an idea how to play the incoming messages with the Web Audio API? i have tried several things but i am not sure if i am doing right.

i look forward to your answers :)

Best regards

mvollrath, 2021-07-28:
I would make sure you're configuring audio_capture to produce raw `wave` data and understand what all of the parameters for `channels`, `depth`, and `sample_rate` are set to.  Then push these buffers to an appropriate Web Audio API streaming interface, after stripping any base64 **or** use `compression: "cbor"` when you subscribe to the audio topic and it should come in as a byte array instead of base64 string.

This may be an example of similar task on the browser side https://www.npmjs.com/package/websockets-streaming-audio

TobiMiller, 2021-07-29:
Thank you very much for your answer! I was already afraid that no one will be interested.

I am aware of channels and sample_rate. But where do I need the depth information in the web audio api?

The Buffer-lenght of 320 is the length of the bytearray converted from base64. The play function is called on every callback. 

i will have a look into your link. thanks.

```var context = new AudioContext();
var buffer = context.createBuffer(1, 320, 16000);

function play(base64) {
  context.resume();
  var binary_string = window.atob(base64);
  var len = binary_string.length;
  var buf = new Float32Array(len);

  //start_iter += 1;
  for (var i = start_iter; i < len + start_iter; i++) {
      var value = binary_string.charCodeAt(i) / 128.0 - 1;
      if (isNaN(value)){
        value = 0;
      }
      nowBuffering[i] = value;
  }
  buffer.copyToChannel(buf, 0);
  var source = context.createBufferSource();
  source.buffer = buffer;
  source.connect(context.destination)
  source.start(0)
}`

jtbandes, 2021-08-19:
Closing this issue since it isn't directly related to the rosbridge_suite, and doesn't require action from the maintainers, but is more a general question of how to interpret audio_common msgs and play them with the Web Audio API. If you still need help I'd recommend answers.ros.org or Stack Overflow.

closed 2021-08-19 by jtbandes