RobotWebTools/rosbridge_suite #577: Renaming develop branch to ros1

jihoonl, 2021-07-29:
I plan to rename develop branch as ros1 to slowly move toward clear visibility for both ros1 and ros2 supports. Please let me know if anyone has concerns about this branch renaming. @RobotWebTools/active @RobotWebTools/authors 

With no concern raises, I will rename `develop` to `ros1` on the first week of August.

amacneil, 2021-07-29:
I support this

flynneva, 2021-07-29:
with this change will the default branch switch then to the new name or will you also switch the default branch? I've noticed a lot of core ros packages switching their default branch to the ROS2 develop or rolling branch.

amacneil, 2021-07-29:
Related: is there any consistency in ROS packages between calling the default branch `main` / `master` / `develop`? I notice a lot of `RobotWebTools` packages use develop, but there doesn't seem to be consistency between other ROS packages I reviewed.

I would advocate for just using `main` for the ros2 branch since it is the new github default.

flynneva, 2021-07-29:
>I would advocate for just using `main` for the ros2 branch

seconded!

jihoonl, 2021-08-10:
Sorry for the late responses guys.

@flynneva 
> with this change will the default branch switch then to the new name or will you also switch the default branch? I've noticed a lot of core ros packages switching their default branch to the ROS2 develop or rolling branch.

the default branch will become `ros2` at some point. But, I would prefer to do it step by step with a slower transition as there are more ros1 users than ros2 as far as I know at this moment. I will definitely change the default branch to `ros2` if I start to see more active development happens in ros2.

@amacneil 
> Related: is there any consistency in ROS packages between calling the default branch main / master / develop? I notice a lot of RobotWebTools packages use develop, but there doesn't seem to be consistency between other ROS packages I reviewed.

It is due to the historic reason as Robot Web Tool has been 10 years around. The default branch of github and ros at that time was `master` and `develop`. Thus, it was just not changing since then. :smile: 

>  I would advocate for just using main for the ros2 branch since it is the new github default (which e.g. nav2 uses).

I would have used `main`  for ros2 if rosbridge is solely for ros2. However, I think using `ros1` and `ros2` separately more makes sense at this moment as the repo supports both releases.

I believe nav2 uses `main` because it is designed for ros2 since its start. If any repo is starting solely for ros2, I definitely agree that using `main` for ros2 support is clear.

How do you think?

amacneil, 2021-08-10:
Makes sense. I think calling the branches `ros1` and `ros2` is fine.

I think it would also be ok to call the `ros2` branch `main`, but maybe calling it `ros2` is more explicit.

amacneil, 2021-08-13:
I have renamed `develop` to `ros1` and updated the default branch to `ros2`.

closed 2021-08-13 by amacneil