RobotWebTools/rosbridge_suite #581: [Build fails] rosapi: targets with same name

ShivamPR21, 2021-08-08:
Hi! all,
I am facing some problems with the build of `rosapi` with `colcon`.

## Command used
```shell
cd <rosbridge_suite_ws>
colcon build
```

## Errors
```shell
--- stderr: rosapi
CMake Error at /home/shivam/ros2_galactic/install/ament_cmake_python/share/ament_cmake_python/cmake/ament_python_install_package.cmake:108 (add_custom_target):
  add_custom_target cannot create target "ament_cmake_python_symlink_rosapi"
  because another target with the same name already exists.  The existing
  target is a custom target created in source directory
  "/home/shivam/autoware/tmp/src/rosbridge_suite/rosapi".  See documentation
  for policy CMP0002 for more details.
Call Stack (most recent call first):
  /home/shivam/ros2_galactic/install/ament_cmake_python/share/ament_cmake_python/cmake/ament_python_install_package.cmake:39 (_ament_cmake_python_install_package)
  /home/shivam/ros2_galactic/install/rosidl_generator_py/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:124 (ament_python_install_package)
  /home/shivam/ros2_galactic/install/ament_cmake_core/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /home/shivam/ros2_galactic/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:292 (ament_execute_extensions)
  CMakeLists.txt:15 (rosidl_generate_interfaces)

---
Failed   <<< rosapi [8.45s, exited with code 1]
```
I am not able to intercept/interpret this problem, I have checked, and there isn't any other previous install for such packages. 

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): rolling/galactic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): Ubuntu 20.04
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): ros2
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): 18.9.0
  - Installation type: Source (ROS & ROSbridge)

ShivamPR21, 2021-08-08:
Even setting [CMP0002](https://cmake.org/cmake/help/v3.19/policy/CMP0002.html) policy to `OLD` doesn't seem to work. 
```cmake
cmake_policy(SET CMP0002 OLD)
```

Also tried setting [ALLOW_DUPLICATE_CUSTOM_TARGETS](https://cmake.org/cmake/help/latest/prop_gbl/ALLOW_DUPLICATE_CUSTOM_TARGETS.html)
```cmake
set_property(GLOBAL PROPERTY ALLOW_DUPLICATE_CUSTOM_TARGETS 1)
```
This also not working, I think these properties have to be set by the core packages.

jtbandes, 2021-08-18:
I also noticed this problem in rolling when we tried to add it to CI, so we'll need to resolve it in order to do this: #598 I don't know why it would fail in rolling but it works for galactic/foxy in CI and on my machine.

jtbandes, 2021-08-19:
This issue is caused by https://github.com/ros2/rosidl_python/pull/131.

closed 2021-10-15 by jtbandes