RobotWebTools/rosbridge_suite #583: Feature Request: Add the ability to filter / limit data returned from "get_topics_and_raw_types" to ROSBridge

gkjohnson, 2021-08-10:
Hello! I originally created an issue over in ROSLibjs [here](https://github.com/RobotWebTools/roslibjs/issues/460) and it was mentioned that this was the better location for the request but ultimately my use case involves ROSLibjs.

I'm not an expert in the ROSBridge codebase but I can try to make PR if anyone can give me some guidance. And if the feature is added to ROSBridge though I can update ROSLibJS to support the new API.

## Expected Behavior

The `getTopicsAndRawTypes` function affords the ability to return all the raw message type descriptions available which is extremely useful but this can result in a _huge_ bundle of data when there are a lot of topics which takes a while to send. It also returns redundant data if multiple topics share a single topic type -- in my case the amount of listed topic types (and therefore message definitions) is more than double what it needs to be and it takes ~13 seconds to receive the message through a localhost connection.

It would be nice if there were a way to filter the topics returned and / or cut down on redundant data included in these messages. Here are a couple API options that come to mind:

1. Add a topics array option to `get_topics_and_raw_types` which limits the topics returned to the ones in the list.
2. Add a function like `get_raw_type` or `get_raw_message_description` to return the just the "raw message description" for a single topic name (similar to `get_topic_type` for topic names) so the app can request only the data it needs.

I'm leaning towards options 2 personally but I defer to the maintainers preference.

## Actual Behavior

All message types are returned in one bundle with redundant data which is not only more than is needed but also significantly slows down socket requests.

## Steps to Reproduce the Problem

  1. Start up rosbridge with a project with many ros topics.
  1. Make a request for "getTopicsAndRawTypes" through ROSLibjs.
  1. Content is huge and takes a long time to return.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): melodic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): bionic
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 0.11.12
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): 17.9.0

jtbandes, 2021-08-10:
Could you share some more info about the size of the system you're working with? How many topics/messages, and what's the total byte size of the API response here?

gkjohnson, 2021-08-10:
We're dealing with a relatively large simulation setup. Here's some information about the returned values when running `getTopicsAndRawTypes` with ROSLibjs:

- The response provides an array of 559 topics (and associated type information)
- Of those 559 returned types only 201 are unique meaning 358 of the returned typedefs are redundant
- From the websocket frames inspector the length of the message from rosbridge is `1327535` (which I believe is in bytes but it's unlabeled in the Chrome UI...) which is a 1.3MB message.

Reducing the redundancy of the typdefs array would be a fairly large improvement on its own (though that would be a breaking change) but even then I will likely only need a small subset of those for my application.

jtbandes, 2021-08-10:
A 1.3MB message should not take 13 seconds to transmit on localhost...maybe the delay is in gathering the datatype information in the Python code?

gkjohnson, 2021-08-10:
> A 1.3MB message should not take 13 seconds to transmit on localhost...

I didn't intend to imply that the 13 seconds was all websocket transfer time but you're right -- I see now when I test it takes around 1 second. I think I had inadvertently included extra delay from something like buffered startup / connection time for the websocket or rosbridge when I did my tests a few weeks ago for the original issue. I'll move forward with the implementation as it is and open a new issue if I find it causes any issues.

Sorry for the noise! 😁

closed 2021-08-10 by gkjohnson