RobotWebTools/rosbridge_suite #610: "ValueError: cannot serialize unknown object" or "IndexError: list index out of range" with mixed raw and non-raw subscribers

jtbandes, 2021-08-19:
## Expected Behavior
Multiple clients should be able to connect with different encoding/compression requirements, such as one using `cbor-raw` and one using `none` (json).

## Actual Behavior
A single MultiSubscriber is created per topic, and the `raw` argument, which affects the type of messages received in the subscriber callback, is set for the whole MultiSubscriber:

https://github.com/RobotWebTools/rosbridge_suite/blob/341294377d93fb91a07dd479e442c6bc5c2c4d05/rosbridge_library/src/rosbridge_library/internal/subscribers.py#L232-L234

Depending on the order of subscriber connections, one json subscriber and one cbor-raw subscriber leads to either

```
[rosbridge_websocket-1] [ERROR] [1629395851.020892139] [rosbridge_websocket]: Exception calling subscribe callback: Traceback (most recent call last):
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/subscribers.py", line 189, in callback
[rosbridge_websocket-1]     callback(outgoing)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 163, in on_msg
[rosbridge_websocket-1]     self.handler.handle_message(msg)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/subscription_modifiers.py", line 70, in handle_message
[rosbridge_websocket-1]     self.publish(msg)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 152, in _publish
[rosbridge_websocket-1]     self.publish(message, self.fragment_size, self.compression)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 333, in publish
[rosbridge_websocket-1]     outgoing_msg = bytearray(encode_cbor(outgoing_msg))
[rosbridge_websocket-1] ValueError: cannot serialize unknown object: std_msgs.msg.Int32MultiArray(layout=std_msgs.msg.MultiArrayLayout(dim=[], data_offset=0), data=[1, 2])
[rosbridge_websocket-1] 
```

 or 

```
[rosbridge_websocket-1] [ERROR] [1629399879.030726331] [rosbridge_websocket]: Exception calling subscribe callback: Traceback (most recent call last):
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/subscribers.py", line 189, in callback
[rosbridge_websocket-1]     callback(outgoing)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 163, in on_msg
[rosbridge_websocket-1]     self.handler.handle_message(msg)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/subscription_modifiers.py", line 70, in handle_message
[rosbridge_websocket-1]     self.publish(msg)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 152, in _publish
[rosbridge_websocket-1]     self.publish(message, self.fragment_size, self.compression)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 335, in publish
[rosbridge_websocket-1]     outgoing_msg["msg"] = message.get_json_values()
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/outgoing_message.py", line 18, in get_json_values
[rosbridge_websocket-1]     self._json_values = extract_json_values(self._message)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/message_conversion.py", line 124, in extract_values
[rosbridge_websocket-1]     rostype = msg_instance_type_repr(inst)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/message_conversion.py", line 145, in msg_instance_type_repr
[rosbridge_websocket-1]     return '{}/{}'.format(inst_repr[0], inst_repr[2].split('(')[0])
[rosbridge_websocket-1] IndexError: list index out of range
```

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): galactic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): focal
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 1.0.7

Interpause, 2022-08-09:
To add on, Foxglove uses `cbor-raw`, which makes it quite difficult for me to code my own dashboard because of this. I have no idea where to even begin decoding `cbor-raw`.

jtbandes, 2023-08-14:
You can also try using the [foxglove_bridge](https://github.com/foxglove/ros-foxglove-bridge). We provide [client libraries](https://github.com/foxglove/ws-protocol) and various examples of how to use them.

closed 2024-09-13 by github-actions[bot]