RobotWebTools/rosbridge_suite #619: Can not install rosbridge-server (Noetic)

trungle1798, 2021-08-20:
## Expected Behavior
Hi all, I tried to install rosbridge follow the command
sudo apt-get install ros-noetic-rosbridge-server 

## Actual Behavior
I kept receiving this type of error
Depends: python3-tornado but it is not installable
I tried to install the package but another bug happened
E: Package 'python3-tornado' has no installation candidate
It seems like there's only pip3-tornado not python3-tornado

## Steps to Reproduce the Problem

  1.sudo apt-get install ros-noetic-rosbridge-server 
  2.sudo apt-get install python3-tornado
 

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): noetic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): Ubuntu 20.04
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): Latest version
  - Tornado Version (`python -c 'import tornado; print tornado.version'`): 6.1

amacneil, 2021-08-20:
I ran `apt-get install ros-noetic-rosbridge-server` and `apt-get install python3-tornado` successfully inside a `ros:noetic-ros-core` docker container.

I think you probably have some apt repos in `/etc/apt/sources.list.d` that are messing things up for you?

trungle1798, 2021-08-20:
> I ran `apt-get install ros-noetic-rosbridge-server` and `apt-get install python3-tornado` successfully inside a `ros:noetic-ros-core` docker container.
> 
> I think you probably have some apt repos in `/etc/apt/sources.list.d` that are messing things up for you?

Yeah, I had to reinstall Ubuntu, and things were back to normal. Thank you for your reply

closed 2021-08-20 by trungle1798

crisdeodates, 2022-01-11:
I am using ROS-noetic. 
On trying:

`$ sudo apt-get install ros-noetic-rosbridge-server`

I get the following output:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-noetic-rosbridge-suite
```

amacneil, 2022-01-13:
@crisdeodates did you follow http://wiki.ros.org/noetic/Installation/Ubuntu ?

What does `cat /etc/apt/sources.list.d/ros*.list` return?

crisdeodates, 2022-01-13:
@amacneil Yes I had followed that tutorial while I installed ROS

```
$ cat /etc/apt/sources.list.d/ros*.list
# deb http://packages.ros.org/ros/ubuntu focal main
```

amacneil, 2022-01-14:
Maybe you need to run `apt update`?

If you already tried that, something is broken on your computer. It works fine in a fresh container:

```sh
$ docker run --rm -it ubuntu:focal bash
root@d4004bc05747:/# apt update
  <...etc...>
root@d4004bc05747:/# apt install curl gnupg
  <...etc...>
root@d4004bc05747:/# echo "deb http://packages.ros.org/ros/ubuntu focal main" > /etc/apt/sources.list.d/ros-latest.list
root@d4004bc05747:/# curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add -
OK
root@d4004bc05747:/# apt-get update
  <...etc...>
root@d4004bc05747:/# apt-get install ros-noetic-rosbridge-server
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  binutils binutils-aarch64-linux-gnu binutils-common bzip2-doc cmake cmake-data cpp cpp-9 cython3 default-libmysqlclient-dev distro-info-data docutils-common
  <...etc...>
0 upgraded, 287 newly installed, 0 to remove and 2 not upgraded.
Need to get 125 MB of archives.
After this operation, 745 MB of additional disk space will be used.
Do you want to continue? [Y/n] ^C
```