RobotWebTools/rosbridge_suite #626: how can I install this packages (ros2 foxy) as a python module, so that I can import the bridge_library in my python code

hamzaadlane, 2021-08-20:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`):
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`):
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`):
  - Tornado Version (`python -c 'import tornado; print tornado.version'`):

jtbandes, 2021-08-20:
Have you tried `colcon build` followed by `source install/setup.bash`?

hamzaadlane, 2021-08-21:
yes I build all these ros packages and I source also my workspace but to install these ros packages as a python modules  I guess, I need a file like setup.py to install it.

amacneil, 2021-08-23:
There will be an apt release of rosbridge in the next sync (Galactic sync is [tomorrow](https://discourse.ros.org/t/new-packages-for-galactic-geochelone-2021-07-16/21418), I'm not sure when Foxy sync is). This should be easier than installing from source.

If you're still having issues, please post on [ROS Answers](https://answers.ros.org/questions/).

closed 2021-08-23 by amacneil