RobotWebTools/rosbridge_suite #633: stderr: rosapi  The Python module 'pytest' was not found, pytests can not be run

GaoGeolone, 2021-08-30:
When I try to compile the whole project with cmd " colcon build " on my win10 OS and ROS2 version of Foxy.

## Actual Behavior
```
[1.254s] root DEBUG Using proactor: IocpProactor
Starting >>> rosbridge_library
Starting >>> rosbridge_msgs
[Processing: rosbridge_library, rosbridge_msgs]
Finished <<< rosbridge_msgs [35.5s]
Finished <<< rosbridge_library [52.6s]
Starting >>> rosapi
[Processing: rosapi]
[Processing: rosapi]
--- stderr: rosapi
CMake Warning at C:/opt/ros/foxy/x64/share/ament_cmake_pytest/cmake/ament_add_pytest_test.cmake:77 (message):
  The Python module 'pytest' was not found, pytests can not be run (e.g.  on
  Ubuntu/Debian install the package 'python3-pytest')
Call Stack (most recent call first):
  CMakeLists.txt:60 (ament_add_pytest_test)

---
Finished <<< rosapi [1min 5s]
Starting >>> rosbridge_server
Finished <<< rosbridge_server [8.80s]
Starting >>> rosbridge_suite
Finished <<< rosbridge_suite [3.16s]

Summary: 5 packages finished [2min 10s]
  1 package had stderr output: rosapi
```
## Steps to Reproduce the Problem

  1. Pull from Origin
  1. colcon build

## Specifications

  - ROS Version (`Foxy`):
  - OS Version (`win10 deployed with Chocolatey`):
  - Rosbridge Version (`1.0.8`):
  - Tornado Version (`ModuleNotFoundError: No module named 'tornado'`):

jtbandes, 2021-08-30:
You need to ensure `ament_cmake_pytest` is installed as well as the pytest python module, e.g. `pip3 install pytest`

GaoGeolone, 2021-08-31:
> 
> 
> You need to ensure `ament_cmake_pytest` is installed as well as the pytest python module, e.g. `pip3 install pytest`

Thank you for replying. And I tried to ensure if pytest  had  been prepared:
`pip3 install pytest
Requirement already satisfied: pytest in c:\opt\ros\foxy\x64\lib\site-packages (6.1.0)
Requirement already satisfied: pluggy<1.0,>=0.12 in c:\opt\ros\foxy\x64\lib\site-packages (from pytest) (0.13.1)
Requirement already satisfied: iniconfig in c:\opt\ros\foxy\x64\lib\site-packages (from pytest) (1.0.1)
Requirement already satisfied: colorama in c:\opt\ros\foxy\x64\lib\site-packages (from pytest) (0.4.3)
Requirement already satisfied: py>=1.8.2 in c:\opt\ros\foxy\x64\lib\site-packages (from pytest) (1.9.0)
Requirement already satisfied: atomicwrites>=1.0 in c:\opt\ros\foxy\x64\lib\site-packages (from pytest) (1.4.0)
Requirement already satisfied: toml in c:\opt\ros\foxy\x64\lib\site-packages (from pytest) (0.10.1)
Requirement already satisfied: packaging in c:\opt\ros\foxy\x64\lib\site-packages (from pytest) (20.4)
Requirement already satisfied: attrs>=17.4.0 in c:\opt\ros\foxy\x64\lib\site-packages (from pytest) (20.2.0)
Requirement already satisfied: six in c:\opt\ros\foxy\x64\lib\site-packages (from packaging->pytest) (1.15.0)
Requirement already satisfied: pyparsing>=2.0.2 in c:\opt\ros\foxy\x64\lib\site-packages (from packaging->pytest) (2.4.7`

As for ament_cmake_pytest, I could find a document placed at C:\opt\ros\foxy\x64\share\ament_cmake_pytest.

If pytest had been prepared, what else could cause failure in compiling?

GaoGeolone, 2021-08-31:
@jtbandes THANK YOU VERY MUCH. 

I have figured out the reason.
Pytest with version 6.1.0 or above has an implicit conflict between pytest-rerunfailures  and _pytest.resultlog, and that's the reason why the stderr occur:
```
--- stderr: rosapi
CMake Warning at C:/opt/ros/foxy/x64/share/ament_cmake_pytest/cmake/ament_add_pytest_test.cmake:77 (message):
The Python module 'pytest' was not found, pytests can not be run (e.g.  on
  Ubuntu/Debian install the package 'python3-pytest')
Call Stack (most recent call first):
 CMakeLists.txt:60 (ament_add_pytest_test)
---`
```
and here is the place 
 ```
 # ensure pytest is available
  ament_has_pytest(has_pytest QUIET PYTHON_EXECUTABLE "${ARG_PYTHON_EXECUTABLE}")
```
The straight way is update pytest-rerunfailures to version 9.1.1
`
pip install pytest-rerunfailures==9.1.1`

 Here is the reference[https://www.cnblogs.com/QiKa/p/14457269.html](url)

GaoGeolone, 2021-08-31:
Since the problem was solved, I will close this issue.

closed 2021-08-31 by GaoGeolone