RobotWebTools/rosbridge_suite #634: launch.substitutions.substitution_failure.SubstitutionFailure: executable 'rosbridge_websocket' not found on the libexec directory

GaoGeolone, 2021-08-31:
I am not familiar with ROS2, but I am trying to understand it. So when I try to launch rosbridge as
 it used to be `roslaunch rosbridge_server rosbridge_websocket.launch`
But, I noticed that the launchfile had changed to rosbridge_websocket_launch.xml, and with ros2, I typed this cmd after call setup.bat:
`ros2 launch rosbridge_server rosbridge_websocket_launch.xml`

## Expected Behavior
If it goes well, I believe the node of rosbridge_websocket will function.

## Actual Behavior
But, I get:
```
[INFO] [launch]: All log files can be found below C:\Users\Administrator\.ros/log\2021-08-31-15-15-29-618970-DESKTOP-B14PKIN-17048
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at c:\opt\ros\foxy\x64\lib\site-packages\launch\launch_service.py:271> exception=SubstitutionFailure("executable 'rosbridge_websocket' not found on the libexec directory 'F:\\04-ROS\\02-projects\\ros2proxy\\install\\rosbridge_server\\lib\\rosbridge_server' ")>
Traceback (most recent call last):
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 2 more times]
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\action.py", line 108, in visit
    return self.execute(context)
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch_ros\actions\node.py", line 426, in execute
    ret = super().execute(context)
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\actions\execute_process.py", line 775, in execute
    self.__expand_substitutions(context)
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\actions\execute_process.py", line 635, in __expand_substitutions
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\actions\execute_process.py", line 635, in <listcomp>
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\utilities\perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\utilities\perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch\launch_context.py", line 184, in perform_substitution
    return substitution.perform(self)
  File "c:\opt\ros\foxy\x64\lib\site-packages\launch_ros\substitutions\executable_in_package.py", line 84, in perform
    raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: executable 'rosbridge_websocket' not found on the libexec directory 'F:\[private project file name]\ros2proxy\install\rosbridge_server\lib\rosbridge_server'
```

## Steps to Reproduce the Problem

  1. Compile the project
  1. call setup.bat
  1. ros2 launch the xml filie

## My analysis
I came back to read CMakeList.txt, but no add_executable(XXX) was used. 
Could this be the reason why no executable file named after node was made?

## Specifications

  - ROS2 Version (`Foxy deployed with Chocolatey`):
  - OS Version (`win10 `):
  - Rosbridge Version (`1.0.8`):
  - Tornado Version (`ModuleNotFoundError: No module named 'tornado'`):

jtbandes, 2021-08-31:
`rosbridge_websocket` is a symlink to `rosbridge_websocket.py`.  It's installed by CMake here: https://github.com/RobotWebTools/rosbridge_suite/blob/fda0957a11278329338c1769d42c8ba3209e007d/rosbridge_server/CMakeLists.txt#L15

However it might not work on Windows because it's a symlink. Maybe you can try using the .py directly?

I'm not sure why this package uses symlinks — possibly historical reasons. We could probably remove them and change the launch file to use the .py scripts directly.

GaoGeolone, 2021-09-01:
@jtbandes THQ! Did you mean that symlink could work well on UNIX, but windows could not work in the same way?
I tried to use cmd "mklink"， but it could not be regarded as an executable file by the ros2 on windows 10.

And I am going to follow the guidance [Creating a launch file](https://docs.ros.org/en/foxy/Tutorials/Launch-Files/Creating-Launch-Files.html?highlight=launch#write-the-launch-file) to make a .py launch script, and I will report soon.

GaoGeolone, 2021-09-06:
I made it!!! WOW!
```
F:\04-ROS\ros2bridge_ws>ros2 run rosbridge_server rosbridge_websocket
registered capabilities (classes):
 - <class 'rosbridge_library.capabilities.call_service.CallService'>
 - <class 'rosbridge_library.capabilities.advertise.Advertise'>
 - <class 'rosbridge_library.capabilities.publish.Publish'>
 - <class 'rosbridge_library.capabilities.subscribe.Subscribe'>
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - <class 'rosbridge_library.capabilities.advertise_service.AdvertiseService'>
 - <class 'rosbridge_library.capabilities.service_response.ServiceResponse'>
 - <class 'rosbridge_library.capabilities.unadvertise_service.UnadvertiseService'>
[INFO] [1630894405.130708700] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9090
```
But the structre has changed a lot, and I will try to diff the whole project.

GaoGeolone, 2021-09-18:
TCP based Communicate on ROS2 (OS: win10) has been solved. https://github.com/RobotWebTools/rosbridge_suite/issues/639#issuecomment-920504652

closed 2021-09-18 by GaoGeolone