RobotWebTools/rosbridge_suite #635: Adding support for binary data in websocket bridge

stefan-pdx, 2021-09-01:
Hello!

I've started using rosbridge_suite's websocket bridge for publishing data from web applications to ROS. This allows for a web-based sensor simulation tool to publish telemetry for downstream ROS nodes -- this can be quite handy for dev/test environments!

Specifically, I'm leverage Babylon.js to render a screen and obtain screenshots for each frame. That PNG data is then encoded into a `sensor_msgs/CompressedImag` message and published to a topic. After several attempts, I've found that I need to encode the `data` field of the message containing an Array of 8-byte values along the lines of:

```js
scene.onAfterRenderObservable.add(() => {
    ScreenshotTools.CreateScreenshot(engine, camera, 200, (data) => {
        const message = new Message({
            header: {
                stamp: { sec, nanosec },
                frame_id: "main"
            },
            format: "png",
            data: Array.from(atob(data.replace("data:image/png;base64,", "")), c => c.charCodeAt(0)), // eww
        });

        topic.publish(message);
    });
});
```

I attempted to use `data: Uint8Array.from(...)`, however that resulted in some rosbridge_suite parsing errors. `Array` works, but it introduces quite a performance bottleneck.

Is it possible/practice to encode binary data (like `Uint8Array`) within messages published over the websocket?

Thank you!

mvollrath, 2021-09-01:
I think the best you can do right now is [encode the data as a base64 string of the UInt8Array](https://stackoverflow.com/q/12710001).  Server to client optionally uses binary messages encoded with CBOR which is very efficient, but that has not been implemented for client to server yet.

jtbandes, 2021-09-01:
Neat, I didn't know that rosbridge_server supports base64 when publishing messages. Is that documented anywhere? Seems like the relevant code is here:

https://github.com/RobotWebTools/rosbridge_suite/blob/0a1a947e96a41c3a974f2985f5dfbc02a232b1c9/rosbridge_library/src/rosbridge_library/internal/message_conversion.py#L219-L223

Maybe this is good enough for the original use case?

To allow sending binary-only messages, we'd have to decide/specify how the server treats binary messages it receives from the clients. CBOR does seem like a good choice since it's already supported in the other direction.

mvollrath, 2021-09-01:
> Is that documented anywhere?

I don't know of any secret documentation stashes, no :D  Not a use case I ever needed, we used native capture and short deadline VP8/RTP to do something similar as systems solution..

> Maybe this is good enough for the original use case?

Base64 is expensive, but not outlandish for small binary blobs.  Full screen 1080p+ png  seems doable, but might need compromises if it's a full motion stream.  Pushing an ArrayBuffer view out to the socket can be orders of magnitude faster for this kind of data.

> To allow sending binary-only messages, we'd have to decide/specify how the server treats binary messages it receives from the clients. CBOR does seem like an obvious choice since it's already supported in the other direction. On the other hand, it would be nice to accept the message content in a binary format that can be directly published to ROS (so CDR for ROS 2).

CDR looks good on paper, seems like it can do everything required.  An even closer-to-the-wire encoding than CBOR-raw would be interesting (CDR-raw?).

jtbandes, 2021-09-01:
Actually, I forgot that the `cbor-raw` subscription encoding already includes the message as a ROS-encoded binary blob (which is CDR for ROS 2). So we could say a `cbor-raw` published message is expected to contain ROS binary data. (I haven't looked to see if there are Python APIs that support publishing a binary blob, but I assume it's possible.)

It might be useful for the protocol to include some kind of negotiation over supported encoding formats as well, for both publishing and subscribing.

closed 2022-03-31 by github-actions[bot]