RobotWebTools/rosbridge_suite #639: Remove TCP + UDP servers

jtbandes, 2021-09-09:
The TCP and UDP servers were not ported from ROS 1 to ROS 2. Since they don't work, we might as well remove the unused code to reduce the amount of maintenance required on this repo.

GaoGeolone, 2021-09-15:
Excuse me, I am still working on TCP server with ros2 on windows 10. And I have an idea, maybe I could start a new project to maintain the tcp part?

jtbandes, 2021-09-15:
If you have made some changes to get the TCP server working for ROS 2, you are welcome to submit a pull request to this repo with the changes. Ideally you can also add a test like the websocket smoke test: https://github.com/RobotWebTools/rosbridge_suite/blob/ros2/rosbridge_server/test/websocket/smoke.test.py (of course the test will need to be very different to work with TCP)

amacneil, 2021-09-15:
Out of curiosity, what is your use case is for rosbridge tcp?

I assume everywhere you could use tcp, you would also be able to use websocket instead?

GaoGeolone, 2021-09-16:
> If you have made some changes to get the TCP server working for ROS 2, you are welcome to submit a pull request to this repo with the changes. Ideally you can also add a test like the websocket [...]

@jtbandes Mainly because the problems I came cross while applying rosbridge_suite on Win10, that symlink cannot be taken as an executable ros-node file, so I change the rosbridge_server and rosapi into a pure python package with their setup.py files. As a consequent, msg and srv have to be moved out of the package with ros-node( rosbridge_server and rosapi  ). That is to say, the structure of the whole project has changed. It seems diffcult to merge into the current project.
Actually, I keep the websocket server as the way it used to serve in rosbridge_suite. Then, I will try to make test file as you have suggested.

GaoGeolone, 2021-09-16:
> 
> 
> Out of curiosity, what is your use case is for rosbridge tcp?
> 
> I assume everywhere you could use tcp, you would also be able to use websocket instead?

Thank you for asking, as shown below, I am trying to communicate with an Unreal Engine Plugin named ROS-Integration ，this plugin of 3rd party defined only a TCP communication channel with Rosbridge( ROS1 version ).  The way to make them work on ROS2, Either adding a websocket channel to that plugin, or polishing the TCP server. As you can see, I choose the former way

[image]

jtbandes, 2021-09-23:
@GaoGeolone Would you be interested in submitting your patches to this repo? If you have fixed the TCP server, it would be nice to keep that in this repo (rather than a fork) so it's easy for other people to find and use.

GaoGeolone, 2021-09-27:
> 
> 
> @GaoGeolone Would you be interested in submitting your patches to this repo? If you have fixed the TCP server, it would be nice to keep that in this repo (rather than a fork) so it's easy for other people to find and use.

Fine, I am glad to make contributions to this repo.

GaoGeolone, 2021-09-27:
But sorry, How can I submit my version to this repo. The relation between these two repos had broken, due to the reconstruction of my project while trying to figure out the way to work on win10.

Maybe U could help me ?
My version of RosBridge[https://github.com/GaoGeolone/ros2bridge_suite](https://github.com/GaoGeolone/ros2bridge_suite)

closed 2021-11-29 by jtbandes