RobotWebTools/rosbridge_suite #642: Reduce CPU load when having several connections

jstiefel, 2021-09-24:
**Description**

I'm subscribing to an image stream in my web interface. I realized that the CPU load is already relatively high (80-90% on my machine) for the webserver process (1920p, 20fps, image_raw/compressed). However, it works fine when just having one connection. But for every user opening a webbrowser, a new webserver process is started at a workload of 80-90% CPU power resulting in a huge delay up to a crash of the server PC.

Is there any solution to this other than not running the image stream over ROS?

- Library Version: 0.11.13
- ROS Version: Noetic
- Platform / OS: Ubuntu 20.04

**Steps To Reproduce**

Start a webserver with some image stream and opening several browser windows.

**Expected Behavior**

No delay even with several connections.

**Actual Behavior**

Each connection starts a new process with a very high load, slowing down the computer. The image stream gets a delay of up to 10 seconds and crashes from time to time. I tested this with 5 connections.

jstiefel, 2021-09-24:
There was a misinterpretation. The CPU load actually comes from the browser.

amacneil, 2021-09-24:
> There was a misinterpretation. The CPU load actually comes from the browser.

To clarify, does that mean the issue can be closed?

jstiefel, 2021-09-24:
> > There was a misinterpretation. The CPU load actually comes from the browser.
> 
> To clarify, does that mean the issue can be closed?

I've just updated my initial post. Issue is still open and I'm happy for hints.

jtbandes, 2021-09-24:
I would recommend first trying the profiling tools built into Chrome (or other browsers). That will hopefully help you understand what the bottleneck is. https://blog.appsignal.com/2020/02/20/effective-profiling-in-google-chrome.html

mvollrath, 2021-09-24:
In noetic, subscribe topics with queue_length of 1 to prevent backlogging.  This will potentially drop messages, so only do it where that is appropriate (e.g. image streams are perfectly appropriate, you don't actually need to draw every frame).

jstiefel, 2021-09-27:
> In noetic, subscribe topics with queue_length of 1 to prevent backlogging. This will potentially drop messages, so only do it where that is appropriate (e.g. image streams are perfectly appropriate, you don't actually need to draw every frame).

It's totally appropriate for my real time control. But I think the subscriber queue_length is already 1 per default. The hint says 0 means "no queueing". I hope this is right, because in ROS 0 means "infinite queueing". Anyway, I set it to 1 now.

https://github.com/RobotWebTools/roslibjs/blob/27b747662a4a72d1191442aaebc47f77d161c5a9/src/core/Topic.js#L25

I don't really understand the hint of the `queue_size` parameter, but is this the publisher queue size? I will set it to 1 as well:
https://github.com/RobotWebTools/roslibjs/blob/27b747662a4a72d1191442aaebc47f77d161c5a9/src/core/Topic.js#L23

How is roslib.js handling the subscriber, publisher and callback queue? I've read that it is event driven and asynchronous. But what is happening asynchronous?

jstiefel, 2021-09-28:
I actually think there is a performance issue with my javascript, mainly the joystick. So this might not be ROS related, if I interpret the performance metrics from Chrome correctly. The red FPS appear during nippleJS joystick movements.
[image]

jstiefel, 2021-09-29:
I think I fixed the performance issues. It seems like rosbridge was not the limiting factor. Sorry for the post at the wrong place, but it helped me anyway to debug. With the help of the Performance tool, I figured out that the handler of the nippleJS joystick, which calls a `innerHTML` function a lot is causing the high CPU load.

closed 2021-09-29 by jstiefel