RobotWebTools/rosbridge_suite #650: rosbridge stops spinning after a service is called

kenji-miyake, 2021-09-25:
**Description**

Since rosbridge stops spinning, it can't receive the response [here](https://github.com/RobotWebTools/rosbridge_suite/blob/bc06a8dc911fd5ca891a5363021b03cf1f055ca4/rosbridge_library/src/rosbridge_library/capabilities/advertise_service.py#L45-L50). [`prewrite_message` function](https://github.com/RobotWebTools/rosbridge_suite/blob/bc06a8dc911fd5ca891a5363021b03cf1f055ca4/rosbridge_server/src/rosbridge_server/websocket_handler.py#L136) seems to be not running.

- Library Version: bc06a8dc911fd5ca891a5363021b03cf1f055ca4 (latest ros2 branch at 25 Sep)
- ROS Version: ROS2 Galactic
- Platform / OS: Ubuntu 20.04

**Steps To Reproduce**

Please see https://github.com/tier4/rosbridge_suite/pull/8.

**Expected Behavior**

- rosbridge can send a service call request to clients.
- rosbridge can receive the response of the service call.

**Actual Behavior**

- rosbridge can't send the request.
- rosbridge can't receive the response.

kenji-miyake, 2021-09-25:
FYI

- If I add `IOLoop.instance().start()` after [here](https://github.com/RobotWebTools/rosbridge_suite/blob/bc06a8dc911fd5ca891a5363021b03cf1f055ca4/rosbridge_server/src/rosbridge_server/websocket_handler.py#L133), it can send the request, but it of course causes another error.
- We quickly modified this [here](https://github.com/tier4/rosbridge_suite/commit/d3414b2cd6e819a4689871b3f2bc30465742df66), but I believe there is a better way to solve this, so I'd like to discuss it.

kenji-miyake, 2021-10-17:
@jtbandes I'm sorry to be late. Yes, it looks good to me and worked for my use case!

Stephenjerry, 2021-11-19:
@jtbandes hi, bro, I found a another issue when I use your <service async> branch, that is when I disconnect my client with rosbridge websocket server, then I reconnect my client to server, there will an error displayed "[rosbridge_websocket]: WebSocketClosedError: Tried to write to a closed websocket" any time when I call a service, so any solution to that??? Thanks so much. I just wondered the previous websocket connection is still be using with closed status. Is that right?

jtbandes, 2021-11-20:
@Stephenjerry that's interesting, sounds like something we could write a test for, would you be able to add the scenario you describe to `advertise_service.test.py`?

Stephenjerry, 2021-11-21:
I just solved it, my buddy, I deleted a line of code, then everything worked ok.  A line of code that should be deleted is "  self.protocol.external_service_list[service_name].service_handle.shutdown(
                "Duplicate advertiser."" in advertise_service.py script.

Stephenjerry, 2021-11-21:
I think this line of code doesn't make any sense, and it seems like blocking the reconnection pipline of same client, so that should be the matter. Do you think so? Or you can try it out like what I described above. Anyway, thank you so much, my bro

Stephenjerry, 2021-11-21:
@jtbandes

closed 2021-11-29 by jtbandes