RobotWebTools/rosbridge_suite #661: Rolling package job for rosapi failing.

nuclearsandwich, 2021-09-29:
ros/rosdistro#30483 released rosbridge_suite into Rolling. However the [rosapi](https://build.ros2.org/view/Rbin_uF64/job/Rbin_uF64__rosapi__ubuntu_focal_amd64__binary) packaging job never successfully built.

- Library Version: 1.0.5
- ROS Version: Rolling
- Platform / OS: Ubuntu 20.04

The build appears to be failing during CMake configuration. An excerpt from the most recent build log as of this writing is below.

```
CMake Error at /opt/ros/rolling/share/ament_cmake_python/cmake/ament_python_install_package.cmake:98 (add_custom_target):
  add_custom_target cannot create target "ament_cmake_python_symlink_rosapi"
  because another target with the same name already exists.  The existing
  target is a custom target created in source directory
  "/tmp/binarydeb/ros-rolling-rosapi-1.0.5".  See documentation for policy
  CMP0002 for more details.
```

nuclearsandwich, 2021-09-29:
FYI @jtbandes, who made the rosbridge_suite release.

jtbandes, 2021-09-29:
Thanks for the report! This is already tracked in #581.

This issue is blocked by https://github.com/ros2/rosidl_python/issues/141. Is rosapi the only package running into this problem?

closed 2021-09-29 by jtbandes