RobotWebTools/rosbridge_suite #688: Can't request message details of common messages

GiteZz, 2021-12-07:
**Description**

When trying to use the RosAPI to request message details I get an exception in the rosapi node. This happens with a common messages included with ROS2. 

- Library Version: Latest commit in ROS2 branch (67442076c90922b39b21c33908e892a3929daff5)
- ROS Version: Galactic
- Platform / OS: Ubuntu 20.04 (Running in Docker)

**Steps To Reproduce**

`ros2 service call /rosapi/message_details rosapi_msgs/srv/MessageDetails "{type: geometry_msgs/Twist}"`

Same happens with the std_msgs:

`ros2 service call /rosapi/message_details rosapi_msgs/srv/MessageDetails "{type: std_msgs/Float32}"`

**Expected Behavior**

Get the message details for the requested message type.

**Actual Behavior**

```
[rosapi_node-2] Traceback (most recent call last):
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/rosapi/rosapi_node", line 322, in <module>
[rosapi_node-2]     main()
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/rosapi/rosapi_node", line 313, in main
[rosapi_node-2]     rclpy.spin(node)
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/__init__.py", line 196, in spin
[rosapi_node-2]     executor.spin_once()
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 707, in spin_once
[rosapi_node-2]     raise handler.exception()
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
[rosapi_node-2]     self._handler.send(None)
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 415, in handler
[rosapi_node-2]     await call_coroutine(entity, arg)
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 369, in _execute_service
[rosapi_node-2]     response = await await_or_execute(srv.callback, request, srv.srv_type.Response())
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 107, in await_or_execute
[rosapi_node-2]     return callback(*args)
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/rosapi/rosapi_node", line 225, in get_message_details
[rosapi_node-2]     dict_to_typedef(d) for d in objectutils.get_typedef_recursive(request.type)
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 98, in get_typedef_recursive
[rosapi_node-2]     return _get_typedefs_recursive(type, [])
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 225, in _get_typedefs_recursive
[rosapi_node-2]     return _get_subtypedefs_recursive(typedef, typesseen)
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 235, in _get_subtypedefs_recursive
[rosapi_node-2]     typedefs = typedefs + _get_typedefs_recursive(fieldtype, typesseen)
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 223, in _get_typedefs_recursive
[rosapi_node-2]     typedef = get_typedef(type)
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 78, in get_typedef
[rosapi_node-2]     return _get_typedef(instance)
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 164, in _get_typedef
[rosapi_node-2]     fieldtypes.append(_type_name(field_type, field_instance))
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 253, in _type_name
[rosapi_node-2]     return _type_name_from_instance(instance)
[rosapi_node-2]   File "/workspaces/mono/ws/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 257, in _type_name_from_instance
[rosapi_node-2]     mod = instance.__module__
[rosapi_node-2] AttributeError: 'float' object has no attribute '__module__'
```

GiteZz, 2021-12-07:
I've done a bit of debugging to help you guys and I think the definition might have changed from float32/float64 to float/double. If I change the atomics in objectutiles.py from 

```
atomics = [
    "bool",
    "byte",
    "int8",
    "uint8",
    "int16",
    "uint16",
    "int32",
    "uint32",
    "int64",
    "uint64",
    "float32",
    "float64",
    "string",
]
```

to 

```
atomics = [
    "bool",
    "byte",
    "int8",
    "uint8",
    "int16",
    "uint16",
    "int32",
    "uint32",
    "int64",
    "uint64",
    "float",
    "double",
    "string",
]
```

my problems disapear

jtbandes, 2021-12-08:
Thanks! Would you be interested in preparing a PR and some tests to fix the issue?

closed 2022-07-07 by github-actions[bot]

scottbell, 2023-01-06:
Just ran into this again with humble. Calling:
`ros2 service call /rosapi/message_details rosapi_msgs/srv/MessageDetails "{type: std_msgs/Float32}"`
causes rosbridge to crash:
```
 $ros2 launch rosbridge_server rosbridge_websocket_launch.xml 
[INFO] [launch]: All log files can be found below /home/scott/.ros/log/2023-01-06-12-05-52-599792-rosmachine-4566
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rosbridge_websocket-1]: process started with pid [4567]
[INFO] [rosapi_node-2]: process started with pid [4569]
[rosbridge_websocket-1] [INFO] [1673003153.749749059] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9090
[rosapi_node-2] Traceback (most recent call last):
[rosapi_node-2]   File "/opt/ros/humble/lib/rosapi/rosapi_node", line 329, in <module>
[rosapi_node-2]     main()
[rosapi_node-2]   File "/opt/ros/humble/lib/rosapi/rosapi_node", line 321, in main
[rosapi_node-2]     rclpy.spin(node)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
[rosapi_node-2]     executor.spin_once()
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 713, in spin_once
[rosapi_node-2]     raise handler.exception()
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[rosapi_node-2]     self._handler.send(None)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
[rosapi_node-2]     await call_coroutine(entity, arg)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 372, in _execute_service
[rosapi_node-2]     response = await await_or_execute(srv.callback, request, srv.srv_type.Response())
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[rosapi_node-2]     return callback(*args)
[rosapi_node-2]   File "/opt/ros/humble/lib/rosapi/rosapi_node", line 228, in get_message_details
[rosapi_node-2]     dict_to_typedef(d) for d in objectutils.get_typedef_recursive(request.type)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 98, in get_typedef_recursive
[rosapi_node-2]     return _get_typedefs_recursive(type, [])
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 223, in _get_typedefs_recursive
[rosapi_node-2]     typedef = get_typedef(type)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 78, in get_typedef
[rosapi_node-2]     return _get_typedef(instance)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 164, in _get_typedef
[rosapi_node-2]     fieldtypes.append(_type_name(field_type, field_instance))
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 253, in _type_name
[rosapi_node-2]     return _type_name_from_instance(instance)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 257, in _type_name_from_instance
[rosapi_node-2]     mod = instance.__module__
[rosapi_node-2] AttributeError: 'float' object has no attribute '__module__'. Did you mean: '__mod__'?
[ERROR] [rosapi_node-2]: process has died [pid 4569, exit code 1, cmd '/opt/ros/humble/lib/rosapi/rosapi_node --ros-args -r __node:=rosapi --params-file /tmp/launch_params_2df8zmzu --params-file /tmp/launch_params_m6j5dj79 --params-file /tmp/launch_params_tg0uwis_'].
```

Using @GiteZz's fix, I was able to get a bit further, but now crashing asking about `rcl_interfaces/msg/ParameterEvent}`:
`ros2 service call /rosapi/message_details rosapi_msgs/srv/MessageDetails "{type: rcl_interfaces/msg/ParameterEvent}"`
```
$ ros2 launch rosbridge_server rosbridge_websocket_launch.xml 
[INFO] [launch]: All log files can be found below /home/scott/.ros/log/2023-01-06-12-11-30-370701-rosmachine-4691
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rosbridge_websocket-1]: process started with pid [4692]
[INFO] [rosapi_node-2]: process started with pid [4694]
[rosbridge_websocket-1] [INFO] [1673003491.543589196] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9090
[rosapi_node-2] Traceback (most recent call last):
[rosapi_node-2]   File "/opt/ros/humble/lib/rosapi/rosapi_node", line 329, in <module>
[rosapi_node-2]     main()
[rosapi_node-2]   File "/opt/ros/humble/lib/rosapi/rosapi_node", line 321, in main
[rosapi_node-2]     rclpy.spin(node)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
[rosapi_node-2]     executor.spin_once()
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 713, in spin_once
[rosapi_node-2]     raise handler.exception()
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[rosapi_node-2]     self._handler.send(None)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
[rosapi_node-2]     await call_coroutine(entity, arg)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 372, in _execute_service
[... cut ...]
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosbridge_library/internal/ros_loader.py", line 213, in _splittype
[rosapi_node-2]     raise InvalidTypeStringException(typestring)
[rosapi_node-2] rosbridge_library.internal.ros_loader.InvalidTypeStringException: octet is not a valid type string
[ERROR] [rosapi_node-2]: process has died [pid 4694, exit code 1, cmd '/opt/ros/humble/lib/rosapi/rosapi_node --ros-args -r __node:=rosapi --params-file /tmp/launch_params_eoskae9r --params-file /tmp/launch_params_oehzpv9e --params-file /tmp/launch_params_nq64wis7'].
```

brag00n, 2023-05-12:
Hello, I have got the same issue using roslibpy API call. E.g.:
`roslibpy.Ros.get_message_details('geometry_msgs/msg/PoseWithCovarianceStamped')`
cause:

`[rosapi_node-2] Traceback (most recent call last):
[rosapi_node-2]   File "/opt/ros/humble/lib/rosapi/rosapi_node", line 329, in <module>
[rosapi_node-2]     main()
[rosapi_node-2]   File "/opt/ros/humble/lib/rosapi/rosapi_node", line 321, in main
[rosapi_node-2]     rclpy.spin(node)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
[rosapi_node-2]     executor.spin_once()
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 713, in spin_once
[rosapi_node-2]     raise handler.exception()
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[rosapi_node-2]     self._handler.send(None)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
[rosapi_node-2]     await call_coroutine(entity, arg)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 372, in _execute_service
[rosapi_node-2]     response = await await_or_execute(srv.callback, request, srv.srv_type.Response())
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[rosapi_node-2]     return callback(*args)
[rosapi_node-2]   File "/opt/ros/humble/lib/rosapi/rosapi_node", line 228, in get_message_details
[rosapi_node-2]     dict_to_typedef(d) for d in objectutils.get_typedef_recursive(request.type)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 98, in get_typedef_recursive
[rosapi_node-2]     return _get_typedefs_recursive(type, [])
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 225, in _get_typedefs_recursive
[rosapi_node-2]     return _get_subtypedefs_recursive(typedef, typesseen)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 235, in _get_subtypedefs_recursive
[rosapi_node-2]     typedefs = typedefs + _get_typedefs_recursive(fieldtype, typesseen)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 223, in _get_typedefs_recursive
[rosapi_node-2]     typedef = get_typedef(type)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 78, in get_typedef
[rosapi_node-2]     return _get_typedef(instance)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 164, in _get_typedef
[rosapi_node-2]     fieldtypes.append(_type_name(field_type, field_instance))
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 253, in _type_name
[rosapi_node-2]     return _type_name_from_instance(instance)
[rosapi_node-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosapi/objectutils.py", line 257, in _type_name_from_instance
[rosapi_node-2]     mod = instance.__module__
[rosapi_node-2] AttributeError: 'numpy.ndarray' object has no attribute '__module__'. Did you mean: '__mod__'?
`

MartinDong, 2023-09-04:
so，the question how to make