RobotWebTools/rosbridge_suite #697: Add ROS2 Actions to rosbridge protocol

Aposhian, 2021-12-23:
Currently the rosbridge protocol only supports topics and services. It would be very helpful to add support for ROS2 actions in the protocol itself.=. Even though actions are technically an abstraction on top of topics and services, similarly services are an abstraction on top of topics, but we still support them in the protocol.

tnagyzambo, 2022-03-22:
I tried to hack something together by sending a service request to the hidden service `/action/name/_action/send_goal` but no luck. From my tinkering it does not look like ROS2 will let you compose your own action client from services and subscribes.

+1 for this to be a feature

DaAwesomeP, 2022-06-27:
Starting to use/find ROS2-only hardware driven by actions, and this has become a blocking issue. Is there anything that can be done to kickstart this?

closed 2023-11-10 by sea-bass