RobotWebTools/rosbridge_suite #698: Multiple clients can't receive subscription information at the same time by best_effort

IDISWJF, 2021-12-27:
**Description**
Is easy to reproduce
Multiple clients can't receive subscription information at the same time by best_effort

- Library Version: ros2
- ROS Version: foxy
- Platform / OS: ubuntu 20.04

**Steps To Reproduce**

1: ros2 topic pub /aaa --qos-reliability best_effort  std_msgs/msg/String "{data: "hhh"}"
2: client : sub /aaa
3: client : sub /aaa

**Expected Behavior**
2: client :  hhh
3: client:   hhh
**Actual Behavior**
2: client :  hhh
3: client:

IDISWJF, 2021-12-28:
Maybe in function subscribe(self, client_id, callback)  rosbridge_library/internal/subscribers.py

Change default qos  to subscribe Best_effort and Reliable infos as Best_effort

jtbandes, 2022-01-03:
This is probably related to https://github.com/RobotWebTools/rosbridge_suite/issues/610

closed 2023-08-05 by github-actions[bot]