RobotWebTools/rosbridge_suite #699: Some message types fail to be forwarded by rosbridge_server to subscribed clients

noirb, 2021-12-30:
**Description**
Some message types appear to be silently dropped by rosbridge_server (using websocket_launch). In particular, `shape_msgs/msg/MeshTriangle` (but there may be other similar messages which cause the same issue).

When a `MeshTriangle` message is published, it can be received by local nodes, but rosbridge client subscribers never receive the message. Additionally, if a message *containing* a list of `MeshTriangle` is published, this message may be received by rosbridge clients if the list is empty, but as soon as the list contains data for at least one `MeshTriangle` the whole message appears to be lost or discarded. (i.e. a `shape_msgs/msg/Mesh` message can be successfully received by a client if the `triangles` field is empty, but not if the field contains at least one entry)

- Library Version: 1.1.1
- ROS Version: ROS2 Galactic
- Platform / OS: Ubuntu 20.04.3

**Steps To Reproduce**: `MeshTriangle` not received by client

```bash
# launch rosbridge
ros2 launch rosbridge_server rosbridge_websocket_launch.xml

# publish a MeshTriangle
ros2 topic pub /test/triangle shape_msgs/msg/MeshTriangle "vertex_indices: [1, 2, 3]"

# echo the topic to verify the publisher is sending data
ros2 topic echo /test/triangle
```

Using the roslibjs simple example page (or any other working rosbridge client), add a subscriber to the triangle topic, load the page and open a browser console:
```js
  var triangle_topic = new ROSLIB.Topic({
	ros: ros,
	name: '/test/triangle',
	messageType: 'shape_msgs/msg/MeshTriangle'
  });
  triangle_topic.subscribe(function(message) {
    console.log('New Triangle: ', message);
  });
```

**Expected Behavior**
MeshTriangle messages are received on the Javascript side.

**Actual Behavior**
No messages are received on the Javascript side. Additionally, no errors appear to be logged by `rosbridge_server`.

**Steps To Reproduce**: `Mesh` not received by client *only if a `MeshTriangle` is present*

```bash
# launch rosbridge
ros2 launch rosbridge_server rosbridge_websocket_launch.xml

# publish a Mesh containing only one vertex and no triangles
ros2 topic pub /test/mesh shape_msgs/msg/Mesh "{triangles: [], vertices: [{x: 1, y: 2, z: 3}]}"

# echo the topic to verify the publisher is sending data
ros2 topic echo /test/mesh
```

Using the roslibjs simple example page (or any other working rosbridge client), add a subscriber to the mesh topic, load the page and open a browser console:
```js
  var mesh_topic = new ROSLIB.Topic({
	ros: ros,
	name: '/test/mesh',
	messageType: 'shape_msgs/msg/Mesh'
  });
  mesh_topic.subscribe(function(message) {
    console.log('New Mesh: ', message);
  });
```

Verify the message is received on the javascript side: `New Mesh:  Object { triangles: [], vertices: (1) […] }`

Now, change the publisher to include one `MeshTriangle` in the `Mesh.triangles` field:
```bash
ros2 topic pub /test/mesh shape_msgs/msg/Mesh "{triangles: [vertex_indices: [1, 2, 3]], vertices: [{x: 1, y: 2, z: 3}]}"
```

**Expected Behavior**
Mesh messages are received in both cases.

**Actual Behavior**
Mesh messages are *only* received if they contain no `MeshTriangle` data (but additional vertex data is fine).

defunctzombie, 2022-01-03:
@noirb Can you show a screenshot indicating that the ros2 topic echo is displaying a message with a MeshTriangle as expected? My initial thought is that the `topic pub` is malformatted when trying to send triangles and vertices.

jtbandes, 2022-01-03:
Thanks for the report.

@kenji-miyake do you think this issue would be related to / fixed by https://github.com/RobotWebTools/rosbridge_suite/pull/692?

kenji-miyake, 2022-01-06:
@jtbandes I'm sorry for my late reply. Actually, it seems a bit different because it uses `uint32`, not `uint8`.

GertVen, 2022-01-07:
I'm getting the same problem with regards to images. I can stream almost anything through the web bridge (test, strings, pointclouds), but the content of image topics seems to be lost somewhere. The data does seem to be echoed correctly using "ros2 topic echo". The subscription to the webbridge is successful, and it even appears to be sending data, I regularly get the message "Sending two parts, [fragment size:" message, but no data is received at the other side of the bridge.

Setup:
-> OS: Ubuntu 20.04 (Using Docker)
-> ROS: Foxy
-> ROS Web Bridge: cloned from ROS2 branch

GertVen, 2022-01-10:
I managed to solve my problem. It turns out ROSBridge was passing the data through, but fragmented. As such ROSLib picked up the data type was "fragment" instead of anything useful, and as such discarded the message.  I fixed my problem by changing the fragment size in the launch file from 10000000 to 99999999. That solved my problem.

Noirb's issue still stands

kenji-miyake, 2022-01-10:
Regarding `MeshTriangle`, I believe fixed in #692.

noirb, 2022-01-31:
@kenji-miyake : Sorry it took me a while to try your fix, but I'm happy to say it looks like it solved my issue :)