RobotWebTools/rosbridge_suite #704: Roslaunch with rosapi launches rosapi twice

mhl787156, 2022-01-20:
**Description**

Hi, I have stumbled across an odd bug where when I try to launch `ros2 launch rosbridge_server rosbridge_websocket_launch.xml`, it launches two rosnodes named `rosapi` and thus throws a warning. It hasn't caused any problems until now where (I think) it is interferring with my connection to Foxglove. 

The only place I've found reference to this is in this old post: https://answers.ros.org/question/344141/ros2-launch-creates-two-nodes-of-same-type/ and also this issue: https://github.com/ros2/launch/issues/411 which links to this issue https://github.com/ros2/geometry2/issues/124 

Indeed removing the `name=rosapi` field of the rosapi node entry within the launchfile appears to solve this problem (locally for me). 

i.e. from:
```
```
to
```
```

- Library Version:
- ROS Version:
- Platform / OS:

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

rob-clarke, 2022-01-24:
I've had a look and it seems to be in the `params.init()` call. I suspect the call to `rclpy.create_node` in [`params.py:73`](https://github.com/RobotWebTools/rosbridge_suite/blob/53985885a824c25846bef58d67adf99d6647ac57/rosapi/src/rosapi/params.py#L73) is picking up the node name remap from the arguments passed to the main `rosapi` script.

rob-clarke, 2022-01-24:
Minimal repro:

```sh
ros2 run rosapi rosapi_node --ros-args --remap __node:=foo
```

NB: Name doesn't matter, even `rosapi` can be used and will cause issue.

This is related: https://answers.ros.org/question/316870/ros2-composition-and-node-names-with-launch-files/

Ugly fix is to replace [params.py:73](https://github.com/RobotWebTools/rosbridge_suite/blob/53985885a824c25846bef58d67adf99d6647ac57/rosapi/src/rosapi/params.py#L73) with:
```py
_node = rclpy.create_node(f"{parent_node_name.split('/')[-1]}_params",use_global_arguments=False)
```

rob-clarke, 2022-01-24:
Actual fix would likely be to actually remove the parameters sub-node as suggested in the comment above that line.

jtbandes, 2022-01-24:
It seems that the name collision happens in Galactic too, but that in Galactic it doesn't break the ability to call services on the rosapi node.

jtbandes, 2022-01-24:
@rob-clarke I actually like the `use_global_arguments=False` idea. This doesn't seem ugly to me — it seems like a fix that directly addresses the problem. Refactoring per the comment might be better but probably more likely to cause new bugs unless a lot of effort is put in to adding comprehensive tests at the same time.

Why did you change the  name to `f"{parent_node_name.split('/')[-1]}_params"` in your proposed fix?

rob-clarke, 2022-02-04:
PR is up. Modified slightly to keep global arguments but explicitly overwrite the name. Subnode name is set to the same as the parent node but with a `_params` suffix to enable multiple instances of the bridge to run.

PR #707 (2022-02-04): Explicitly rename param subnode
**Public API Changes**
None(ish)

The param subnode is now named `{parent_node_name}_params` when the rosapi node is renamed. Previously it would have the same name.

**Description**
Previously, without any renaming, running the `rosapi` node would spawn a `rosapi_params` subnode. If the top-level node was renamed (or had a name set by a launch file) the param subnode ended up with the same name as the top-level node. This fixes the behaviour by explicitly setting the name of the subnode to the same as the top-level node with `_params` as a suffix.

Unlike the s [... cut]
commit c176a0ba (2022-02-04): Explicitly rename param subnode (#707)

**Public API Changes**
None(ish)

The param subnode is now named `{parent_node_name}_params` when the rosapi node is renamed. Previously it would have the same name.

**Description**
Previously, without any renaming, running the `rosapi` node would spawn a `rosapi_params` subnode. If the top-level node was renamed (or had a name set by a launch file) the param subnode ended up with the same name as the top-level node. This fixes the behaviour by explicitly setting the name of the subnode to the same as the top-level node with `_params` as a suffix.

Unli [... cut]
closed 2022-02-04 by jtbandes