RobotWebTools/rosbridge_suite #721: ros1 RosbridgeTcpSocket won't close and spins in an endless loop

curiosus42, 2022-03-08:
**Description**
When running our ROS application for a longer period of time, we ran into performance issues. Longer search showed that rosbridge started many threads over time, which load the CPU core with 100%.

- Library Version: ros1
- ROS Version: noetic
- Platform / OS: Ubuntu 20.04.3 LTS

**Steps To Reproduce**
1. Start rosbridge_server in tcp mode
2. Connect to the rosbridge tcp socket (no service call or anything is needed)
3. Close socket

**Expected Behavior**
```
[INFO] [1646733954.253789]: [Client 0] connected. 1 client total.
[INFO] [1646733959.259695]: [Client 0] disconnected. 0 client total.
```

**Actual Behavior**
Client stays connected and the cpu fan spins up. htop shows a rosbdige_tcp thread created for the client with a cpu load of 100%.

amacneil, 2022-03-15:
Fixed by https://github.com/RobotWebTools/rosbridge_suite/pull/722

closed 2022-03-15 by amacneil

ribes4, 2022-04-08:
Hi, this problem also happens to me in:

- Library Version: ros1
- ROS Version: noetic
- Platform / OS: Ubuntu 20.04.4 LTS

I'm currently using the release candidate `ros-noetic-rosbridge-server`: **0.11.13-1focal.20220107.015434**

Is there any plan to release these changes and update the release candidate package for ros 1 to fix this bug?

Thank you!

PR #722 (2022-03-15): Closes RobotWebTools/rosbridge_suite#721
**Public API Changes**
None

**Description**

When the TCP connection of the client is closed, the Python Socket.recv(...) function returns an empty byte array. However, since the while loop checks for an empty string, the thread hangs in an infinite loop and causes the connection not to be disconnected.

Also added minor description changes for more clarity.

Fixes #721