RobotWebTools/rosbridge_suite #727: rosbridge_server error using parameter topic_glob in ros2

charlielito, 2022-04-07:
Trying to run the `rosbridge_server` with the `topic_glob` argument raises an error in ros2 galactic

- Library Version:  1.1.2-1focal.20220211.111228
- ROS Version: Galactic
- Platform / OS: Ubuntu 20.04

**Steps To Reproduce**
```
ros2 launch rosbridge_server rosbridge_websocket_launch.xml topics_glob:="[/rosout, /camera/rgb/*]"
```

and the error

```
[INFO] [launch]: All log files can be found below /home/ada/.ros/log/2022-04-07-03-32-47-149184-pop-os-7165
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rosbridge_websocket-1]: process started with pid [7170]
[INFO] [rosapi_node-2]: process started with pid [7172]
[rosapi_node-2] Traceback (most recent call last):
[rosapi_node-2]   File "/opt/ros/galactic/lib/rosapi/rosapi_node", line 322, in <module>
[rosapi_node-2]     main()
[rosapi_node-2]   File "/opt/ros/galactic/lib/rosapi/rosapi_node", line 311, in main
[rosapi_node-2]     node = Rosapi()
[rosapi_node-2]   File "/opt/ros/galactic/lib/rosapi/rosapi_node", line 75, in __init__
[rosapi_node-2]     self.declare_parameter("topics_glob", "[*]")
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 360, in declare_parameter
[rosapi_node-2]     return self.declare_parameters('', [args], ignore_override)[0]
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 487, in declare_parameters
[rosapi_node-2]     self._set_parameters(
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 732, in _set_parameters
[rosapi_node-2]     raise InvalidParameterTypeException(
[rosapi_node-2] rclpy.exceptions.InvalidParameterTypeException: Trying to set parameter 'topics_glob' to '['/rosout', '/camera/rgb/*']' of type 'STRING_ARRAY', expecting type 'STRING'
```
**Expected Behavior**
Run without crashiing

**Actual Behavior**
It crashes

Interpause, 2022-07-03:
I got two errors while trying to set `topics_glob="[*]"` that might reveal what is going on:

```sh
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 4 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 128, in evaluate_parameter_dict
    evaluated_value = value.evaluate(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/parameter_descriptions.py", line 83, in evaluate
    self.__evaluated_parameter_value = perform_typed_substitution(
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/type_utils.py", line 547, in perform_typed_substitution
    return coerce_to_type(
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/type_utils.py", line 254, in coerce_to_type
    return convert_as_yaml(value, f"Failed to convert '{value}' using yaml rules")
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/type_utils.py", line 243, in convert_as_yaml
    raise ValueError(f'{error_msg}: yaml.safe_load() failed\n{err}')
ValueError: Failed to convert '[*]' using yaml rules: yaml.safe_load() failed
```

I then tried `["*"]`:

```sh
[rosbridge_websocket-1] Traceback (most recent call last):
[rosbridge_websocket-1]   File "/opt/ros/foxy/lib/rosbridge_server/rosbridge_websocket", line 332, in <module>
[rosbridge_websocket-1]     main()
[rosbridge_websocket-1]   File "/opt/ros/foxy/lib/rosbridge_server/rosbridge_websocket", line 317, in main
[rosbridge_websocket-1]     node = RosbridgeWebsocketNode()
[rosbridge_websocket-1]   File "/opt/ros/foxy/lib/rosbridge_server/rosbridge_websocket", line 74, in __init__
[rosbridge_websocket-1]     self.protocol_parameter_handling()
[rosbridge_websocket-1]   File "/opt/ros/foxy/lib/rosbridge_server/rosbridge_websocket", line 208, in protocol_parameter_handling
[rosbridge_websocket-1]     for element in topics_glob[1:-1].split(",")
[rosbridge_websocket-1] AttributeError: 'list' object has no attribute 'split'
```

Finally, using `\[*\]` worked. From above errors, it seems `rosapi` or `roslaunch` attempts to use `yaml` to parse `topic_glob`. Afterwards, a rather dubious `topics_glob[1:-1]` is used to remove the square brackets to split out the paths. I think this should be fixed.

charlielito, 2023-07-13:
Waiting for response...

closed 2024-08-14 by github-actions[bot]

howking111, 2024-11-15:
Just like this: topics_glob:=/dummy1,/hintonbot00/cloud_node/msg,/hintonbot00/map,/dummy2