RobotWebTools/rosbridge_suite #729: rosbridge_server does not support "*.idl"-message specifications

StahlTim, 2022-04-12:
When trying to forward custom *.idl messages via the rosbridge_server, each iteration triggers an error (here shown for an [autoware_auto_planning_msgs](https://gitlab.com/autowarefoundation/autoware.auto/autoware_auto_msgs/-/blob/master/autoware_auto_planning_msgs/msg/Trajectory.idl) ):
`[rosapi_node-2] Error processing 'module autoware_auto_planning_msgs {' of 'autoware_auto_planning_msgs/Trajectory': ''module' is an invalid message name. It should have the pattern '^[A-Z][A-Za-z0-9]*$''`

- Library Version: 1.1.2 
- ROS Version: galactic
- Platform / OS: ubuntu 20.04

**Steps To Reproduce**
- Using the definition of autoware auto messages (see at the top of this issue)
- Launching a rosbridge_sever via: `ros2 launch rosbridge_server rosbridge_websocket_launch.xml`

Are *.idl messages generally excluded from rosbridge support, or is there a way to work with them?

jtbandes, 2022-04-12:
I'm not aware of any special support for this in rosbridge. The error message you're getting looks like it is trying to parse the .idl as if it were a .msg file. Do you have a stack trace for the error? And is there a strong reason that these messages are written as custom .idl files rather than custom .msg?

StahlTim, 2022-04-12:
Okay, thanks for the info!

In this case we can probably close this issue. However, it might be helpful to include this info in the Readme/Docs (unless it already exists and I haven't found it).

jtbandes, 2023-04-13:
FYI, nowadays the [foxglove_bridge](https://github.com/foxglove/ros-foxglove-bridge) supports `.idl` definitions.

closed 2023-04-13 by jtbandes