RobotWebTools/rosbridge_suite #738: rosbridge_server launch failed

chenjunnn, 2022-05-08:
**Description**

- Library Version: 1.1.2-1focal.20220430.151453
- ROS Version: Galactic
- Platform / OS: Jetson Nano / Using Docker image `ros:galactic-ros-base`

**Steps To Reproduce**

    ros2 launch rosbridge_server rosbridge_websocket_launch.xml

**Actual Behavior**

```log
➜   ros2 launch rosbridge_server rosbridge_websocket_launch.xml        
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-05-08-06-04-32-520780-ucar-mini-1967
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rosbridge_websocket-1]: process started with pid [2008]
[INFO] [rosapi_node-2]: process started with pid [2010]
[rosapi_node-2] [WARN] [1651989874.405045363] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rosapi_node-2] Traceback (most recent call last):
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
[rosapi_node-2]     return importlib.import_module(module_name, package=pkg_name)
[rosapi_node-2]   File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
[rosapi_node-2]     return _bootstrap._gcd_import(name[level:], package, level)
[rosapi_node-2]   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
[rosapi_node-2]   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
[rosapi_node-2]   File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
[rosapi_node-2]   File "<frozen importlib._bootstrap>", line 657, in _load_unlocked
[rosapi_node-2]   File "<frozen importlib._bootstrap>", line 556, in module_from_spec
[rosapi_node-2]   File "<frozen importlib._bootstrap_external>", line 1166, in create_module
[rosapi_node-2]   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
[rosapi_node-2] ImportError: /opt/ros/galactic/lib/librosapi_msgs__rosidl_generator_c.so: undefined symbol: rosidl_runtime_c__String__copy
[rosapi_node-2] 
[rosapi_node-2] During handling of the above exception, another exception occurred:
[rosapi_node-2] 
[rosapi_node-2] Traceback (most recent call last):
[rosapi_node-2]   File "/opt/ros/galactic/lib/rosapi/rosapi_node", line 322, in <module>
[rosapi_node-2]     main()
[rosapi_node-2]   File "/opt/ros/galactic/lib/rosapi/rosapi_node", line 311, in main
[rosapi_node-2]     node = Rosapi()
[rosapi_node-2]   File "/opt/ros/galactic/lib/rosapi/rosapi_node", line 79, in __init__
[rosapi_node-2]     self.register_services()
[rosapi_node-2]   File "/opt/ros/galactic/lib/rosapi/rosapi_node", line 89, in register_services
[rosapi_node-2]     self.create_service(Topics, "/rosapi/topics", self.get_topics)
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 1452, in create_service
[rosapi_node-2]     check_is_valid_srv_type(srv_type)
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/type_support.py", line 51, in check_is_valid_srv_type
[rosapi_node-2]     check_for_type_support(srv_type)
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
[rosapi_node-2]     msg_or_srv_type.__class__.__import_type_support__()
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rosapi_msgs/srv/_topics.py", line 274, in __import_type_support__
[rosapi_node-2]     module = import_type_support('rosapi_msgs')
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
[rosapi_node-2]     raise UnsupportedTypeSupport(pkg_name)
[rosapi_node-2] rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'rosapi_msgs'
[rosbridge_websocket-1] registered capabilities (classes):
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.call_service.CallService'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.advertise.Advertise'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.publish.Publish'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.subscribe.Subscribe'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.advertise_service.AdvertiseService'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.service_response.ServiceResponse'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.unadvertise_service.UnadvertiseService'>
[rosbridge_websocket-1] Traceback (most recent call last):
[rosbridge_websocket-1]   File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
[rosbridge_websocket-1]     return importlib.import_module(module_name, package=pkg_name)
[rosbridge_websocket-1]   File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
[rosbridge_websocket-1]     return _bootstrap._gcd_import(name[level:], package, level)
[rosbridge_websocket-1]   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
[... 8 lines cut ...]
[rosbridge_websocket-1] During handling of the above exception, another exception occurred:
[rosbridge_websocket-1] 
[rosbridge_websocket-1] Traceback (most recent call last):
[rosbridge_websocket-1]   File "/opt/ros/galactic/lib/rosbridge_server/rosbridge_websocket", line 335, in <module>
[rosbridge_websocket-1]     main()
[rosbridge_websocket-1]   File "/opt/ros/galactic/lib/rosbridge_server/rosbridge_websocket", line 320, in main
[rosbridge_websocket-1]     node = RosbridgeWebsocketNode()
[rosbridge_websocket-1]   File "/opt/ros/galactic/lib/rosbridge_server/rosbridge_websocket", line 121, in __init__
[rosbridge_websocket-1]     RosbridgeWebSocket.client_manager = ClientManager(self)
[rosbridge_websocket-1]   File "/opt/ros/galactic/lib/python3.8/site-packages/rosbridge_server/client_mananger.py", line 54, in __init__
[rosbridge_websocket-1]     self._conn_clients_pub = node_handle.create_publisher(
[rosbridge_websocket-1]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 1282, in create_publisher
[rosbridge_websocket-1]     check_is_valid_msg_type(msg_type)
[rosbridge_websocket-1]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/type_support.py", line 35, in check_is_valid_msg_type
[rosbridge_websocket-1]     check_for_type_support(msg_type)
[rosbridge_websocket-1]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
[rosbridge_websocket-1]     msg_or_srv_type.__class__.__import_type_support__()
[rosbridge_websocket-1]   File "/opt/ros/galactic/lib/python3.8/site-packages/rosbridge_msgs/msg/_connected_clients.py", line 27, in __import_type_support__
[rosbridge_websocket-1]     module = import_type_support('rosbridge_msgs')
[rosbridge_websocket-1]   File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
[rosbridge_websocket-1]     raise UnsupportedTypeSupport(pkg_name)
[rosbridge_websocket-1] rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'rosbridge_msgs'
[ERROR] [rosbridge_websocket-1]: process has died [pid 2008, exit code 1, cmd '/opt/ros/galactic/lib/rosbridge_server/rosbridge_websocket --ros-args -r __node:=rosbridge_websocket --params-file /tmp/launch_params_flif5flf --params-file /tmp/launch_params_jjjduci_ --params-file /tmp/launch_params_qm8u_7yr --params-file /tmp/launch_params_81rjhmu1 --params-file /tmp/launch_params_zz4t2iu1 --params-file /tmp/launch_params_2ptexfso --params-file /tmp/launch_params_mjyfnikr --params-file /tmp/launch_params_538t8f2t --params-file /tmp/launch_params_0w0kxu92 --params-file /tmp/launch_params_h7e9yui6 --params-file /tmp/launch_params_q1mmhxci --params-file /tmp/launch_params_awmiohox'].
[ERROR] [rosapi_node-2]: process has died [pid 2010, exit code 1, cmd '/opt/ros/galactic/lib/rosapi/rosapi_node --ros-args -r __node:=rosapi --params-file /tmp/launch_params_7sqk1om6 --params-file /tmp/launch_params_1lxp_ehr --params-file /tmp/launch_params_xl8jqfc3'].
```

brennand, 2022-05-09:
I also had this problem.  If I clone the head into my workspace and building, the error goes away.

chenjunnn, 2022-05-16:
According to https://github.com/ros2/rosidl/issues/680#issuecomment-1125277312, `apt dist-upgrade` can fix this issue

closed 2022-05-16 by chenjunnn

Roboprotos, 2022-11-24:
Hello, I have the exact same error with the following configuration : 
- Ubuntu 20.04 running on a Jetson Xavier (I'm not using docker)
- ROS2 Galactic installed from binary
- rosbridge_suite installed with : `sudo apt install ros-galactic-rosbridge-suite`

However doing `sudo apt update` and `sudo apt dist-upgrade` doesn't solve the issue.
I have a similar issues of ImportError and undefined symbol with other packages using python.

Anyone knows how to solve that ?
Is it possible to reopen that issue ?