RobotWebTools/rosbridge_suite #750: If you disconnect from the Wi-Fi network and reconnect, the subscription will not work after a while.

vusdo, 2022-06-02:
**Description**
If you disconnect from the Wi-Fi network and reconnect, the subscription will not work after a while.

- Library Version: 0.11.13
- ROS Version: melodic
- Platform / OS: Ubuntu 18.04

**Steps To Reproduce**

1. Write publisher and subscription code in your browser.
2. Subscription data output
3. Disconnect from network
4. Reconnect the network
5. Subscription data does not come in after a while.

etc
  - Publishing is possible.
  - There are topics for which data is output when you subscribe, and there are topics that do not.
  - Data comes in if you do rostopic echo topic.
  - The data that is not actually subscribed is in the form of a string, and the length of the string is 100. Data is received 6 times per second. 6 * 100 = 600 ex) "status|1|90|active|23,12,23|0||1"
  - No "client disconnected" counter in logs when disconnecting from network.
  - Closing the browser results in "client disconnected" in the log.

It seems to affect other clients when the network is disconnected. Any solution?

rms@rms:~$ rosnode info /server/rosbridge_websocket 
--------------------------------------------------------------------------------
Node [/server/rosbridge_websocket]
Publications: 
 * /charge_state [std_msgs/String]
 * /ev_state [std_msgs/String]
 * /rosout [rosgraph_msgs/Log]
 * /server/client_count [std_msgs/Int32]
 * /server/connected_clients [rosbridge_msgs/ConnectedClients]
 * /server/dev_state [std_msgs/String]
 * /server/mqtt [std_msgs/String]
 * /state [std_msgs/String]

Subscriptions: 
 * /charge_state [std_msgs/String]
 * /restart [unknown type]
 * /server/dev_state [std_msgs/String]
 * /server/mqtt [std_msgs/String]
 * /state [std_msgs/String]

Services: 
 * /server/dev_cmd
 * /server/fms_service
 * /server/rosbridge_websocket/get_loggers
 * /server/rosbridge_websocket/set_logger_level
 * /service/io_call

contacting node http://fms:40837/ ...
Pid: 2834
Connections:
 * topic: /charge_state
    * to: /server/rosbridge_websocket
    * direction: outbound (40317 - 127.0.0.1:56502) [47]
    * transport: TCPROS
 * topic: /server/mqtt
    * to: /server/rosbridge_websocket
    * direction: outbound (40317 - 127.0.0.1:42382) [29]
    * transport: TCPROS
 * topic: /rosout
    * to: /rosout
    * direction: outbound (40317 - 127.0.0.1:42192) [13]
    * transport: TCPROS
 * topic: /state
    * to: /server/rosbridge_websocket
    * direction: outbound (40317 - 127.0.0.1:56508) [53]
    * transport: TCPROS
 * topic: /server/dev_state
    * to: /server/rosbridge_websocket
    * direction: outbound (40317 - 127.0.0.1:42318) [20]
    * transport: TCPROS
 * topic: /charge_state
    * to: /server/rosbridge_websocket (http://fms:40837/)
    * direction: inbound
    * transport: TCPROS
 * topic: /server/mqtt
    * to: /server/rosbridge_websocket (http://fms:40837/)
    * direction: inbound
    * transport: TCPROS
 * topic: /state
    * to: /server/rosbridge_websocket (http://fms:40837/)
    * direction: inbound
    * transport: TCPROS
 * topic: /server/dev_state
    * to: /server/rosbridge_websocket (http://fms:40837/)
    * direction: inbound
    * transport: TCPROS

vusdo, 2022-07-28:
please help me

chilatiao, 2023-01-06:
Have you solved this problem? I have the same problem

ilikt, 2023-10-08:
+ 1

closed 2024-11-08 by github-actions[bot]