RobotWebTools/rosbridge_suite #757: [Feature Request] WebRTC via Python aiortc package

Interpause, 2022-06-22:
WebRTC is more efficient than mjpeg video streams and works well for low-bandwidth situations due to its ability to dynamically adjust video quality. Furthermore, WebRTC's DataChannel has a lower latency than Websockets as it uses UDP instead of TCP.

In my own project, I have successfully used [`aiortc`](https://github.com/aiortc/aiortc) for WebRTC. I used a [Python implementation of Socket.io](https://python-socketio.readthedocs.io/en/latest/index.html) as my ICE server. I was sending a camera feed at 1080p as well as human pose data at 30 fps. See here: <https://github.com/Interpause/nicepipe/blob/fd693a6638f8f08549b6a13a93d71b6bea4d6810/nicepipe/output/sio_stream.py> (sorry for messy code).

I might be using ROS soon for my next project, so if I might make a pull request if I end up implementing it using rosbridge.

defunctzombie, 2022-06-30:
We don't have the bandwidth to add webrtc transports right now but if you are keen in doing this we can consider architectures and PR if you do get around to it.

Interpause, 2022-07-01:
yup okay. Actually my plan is to use rosbridge for webrtc signaling and sending data to the frontend. While websockets are probably sufficient, I will experiment with datachannels when I do have time.

closed 2022-07-01 by jtbandes