RobotWebTools/rosbridge_suite #758: Not working with multiple clients, KeyError and tornado.websocket.WebSocketClosedError

Alqio, 2022-06-27:
**Description**
When connecting with two clients, rosbridge cannot maintain connection to both clients at once.

- Library Version: branch ros2, commit 4bcf816726f0402675d54695674ea77fffc2f785
- ROS Version: galactic
- Platform / OS: Ubuntu 20.04, kernel 5.14.0-1042-oem, x86_64. Running rosbridge on docker with image osrf/ros:galactic-desktop. Using JavaScript ROSLIB 1.3.0

**Steps To Reproduce**

1. Start rosbridge and other possible ros nodes
2. Open the same web application in two tabs, connect both to ros, subscribe both to same topic that is published with Durability.TRANSIENT_LOCAL.
3. Refresh web application 2 (the one that has not received data)
4. Make an action from web application 2 that creates a new message to the topic

**Expected Behavior**
Both web applications should receive data from the topic when subscribing and would continue to receive data from the topic.

**Actual Behavior**
Only the first web application receives data from the topic until the second web application updates creates an action that produces a new message to the topic.

On step 3, rosbridge outputs the following error:
```
 [INFO] [1656319136.483727487] [rosbridge_websocket]: Client disconnected. 1 clients total.
 Exception in thread Thread-8:
 Traceback (most recent call last):
   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
     self.run()
   File "/app/ros2_ws/install/rosbridge_server/lib/python3.8/site-packages/rosbridge_server/websocket_handler.py", line 113, in run
     self.protocol.finish()
   File "/app/ros2_ws/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/protocol.py", line 284, in finish
     capability.finish()
   File "/app/ros2_ws/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 378, in finish
     subscription.unregister()
   File "/app/ros2_ws/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 85, in unregister
     manager.unsubscribe(self.client_id, self.topic)
   File "/app/ros2_ws/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/subscribers.py", line 278, in unsubscribe
     self._subscribers[topic].unsubscribe(client_id)
   File "/app/ros2_ws/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/subscribers.py", line 185, in unsubscribe
     del self.subscriptions[client_id]
 KeyError: UUID('9207542e-4af6-4835-b38e-cce2907454db')
 [INFO] [1656319136.587913705] [rosbridge_websocket]: Client connected. 2 clients total.
```

On step 4, rosbridge outputs the following error:
```
 [WARN] [1656319244.473545744] [rosbridge_websocket]: WebSocketClosedError: Tried to write to a closed websocket
 ERROR:tornado.application:Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0x7f3f9e66a220>>, <Future finished exception=WebSocketClosedError()>)
 Traceback (most recent call last):
   File "/usr/local/lib/python3.8/dist-packages/tornado/ioloop.py", line 741, in _run_callback
     ret = callback()
   File "/usr/local/lib/python3.8/dist-packages/tornado/ioloop.py", line 765, in _discard_future_result
     future.result()
   File "/usr/local/lib/python3.8/dist-packages/tornado/gen.py", line 234, in wrapper
     yielded = ctx_run(next, result)
   File "/app/ros2_ws/install/rosbridge_server/lib/python3.8/site-packages/rosbridge_server/websocket_handler.py", line 197, in prewrite_message
     future = self.write_message(message, binary)
   File "/usr/local/lib/python3.8/dist-packages/tornado/websocket.py", line 337, in write_message
     raise WebSocketClosedError()
 tornado.websocket.WebSocketClosedError
 ERROR:tornado.application:Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0x7f3f9e66a220>>, <Future finished exception=WebSocketClosedError()>)
 Traceback (most recent call last):
   File "/usr/local/lib/python3.8/dist-packages/tornado/ioloop.py", line 741, in _run_callback
     ret = callback()
   File "/usr/local/lib/python3.8/dist-packages/tornado/ioloop.py", line 765, in _discard_future_result
     future.result()
   File "/usr/local/lib/python3.8/dist-packages/tornado/gen.py", line 234, in wrapper
     yielded = ctx_run(next, result)
   File "/app/ros2_ws/install/rosbridge_server/lib/python3.8/site-packages/rosbridge_server/websocket_handler.py", line 197, in prewrite_message
     future = self.write_message(message, binary)
   File "/usr/local/lib/python3.8/dist-packages/tornado/websocket.py", line 337, in write_message
     raise WebSocketClosedError()
 tornado.websocket.WebSocketClosedError
```

zflat, 2022-07-13:
@Alqio Could you try the branch in https://github.com/RobotWebTools/rosbridge_suite/pull/741? When you refresh it may be the same issue I have seen in https://github.com/RobotWebTools/rosbridge_suite/issues/744.

Alqio, 2022-07-25:
@zflat Awesome, https://github.com/RobotWebTools/rosbridge_suite/pull/741 seems to fix this issue!

PR #803 (2022-10-17): Improve robustness for multiple client connections
**Public API Changes**
None

**Description**
Supersedes #741
Fixes #783
Fixes #758

This PR improves multi-client connection robustness by 
- avoiding unregistering of subscribers if one client disconnects while all others are still waiting for the [initial callback](https://github.com/RobotWebTools/rosbridge_suite/blob/87b3bf0a1a69a9bcfad5f32cb75303881607a5f1/rosbridge_library/src/rosbridge_library/internal/subscribers.py#L222-L238)
- avoiding a deadlock that could occur when new clients connect, subscribe to a topic and then immediately disc [... cut]
closed 2022-10-17 by achim-k