RobotWebTools/rosbridge_suite #766: Rosbridge_suite not finding package after migration from ros2-web-bridge

Moise2901, 2022-07-21:
**Description**
Hello everyone,

I'm having trouble getting rosbridge_suite to work after migrating from ros2-web-bridge. I currently have a system where there are ROS nodes running on a local laptop and on another device on the same local network. We have setup DDS and have confirmed that both machines can see the other one's nodes and subscribe to either one's topics. There is also a web UI running on the local laptop which allows us to set some parameters on every node. Everything was working when running with ros2-web-bridge, but now that we switched to rosbridge_suite, I'm getting some errors suggesting that the ROS package defined on the other machine does not exist, see the following error:

```
[rosbridge_websocket]: [Client e4412291-aefd-4b34-8a3e-1fca8141af1e] [id: call_service:/my_service:11] call_service InvalidModuleException: Unable to import my_package.srv from package my_package. Caused by: No module named 'my_package'

``` 
I'm wondering if I'm maybe missing a step or a setting to get it up and running. To migrate from ros2-web-bridge, I simply deleted the ros2-web-bridge repo and then installed rosbridge_suite with `sudo apt-get install ros-foxy-rosbridge-server`. Then, after launching my ROS nodes, i run `ros2 launch rosbridge_server rosbridge_websocket_launch.xml` to launch the bridge (I launch the UI at the same time).
- Library Version: 1.2.0
- ROS Version: Foxy
- Platform / OS: Ubuntu 20.04

**Expected Behavior**
Able to call any services (offered by local or remote nodes) from the UI.

**Actual Behavior**
Error message sying my remote package does not exist.

zflat, 2022-07-25:
It sounds like you need to source the `setup.bash` file for the workspace that contains your custom `my_package` messages before you run the `rosbridge_server`. This is because the python interpreter needs to be able to find your custom messages.

Moise2901, 2022-07-25:
Thanks for your answer @zflat !

I forgot to mention this is already done before running the bridge on the local laptop. However, we are sourcing the `setup.sh` file instead of  `setup.bash`, might this be an issue? What confuses me is that when we were running with ros2-web-bridge, this issue wasn't there and nothing changed apart from the bridge. I have read the troubleshooting readme of this repo and found out that the `Tornado` version might be a problem. We are currently running on 5.1.1, is this a problem ?

zflat, 2022-07-27:
It still seems like your python path does not know about your custom messages package. Can you try this?
* Get a python interactive session with this: `python3`
* Inside the interactive session run:
  ```
  import my_package.srv
  ```
  * Substitute the real name of your package

If the above works, then maybe you can share the "type" argument that is being used by the client code for calling the service. Maybe the type argument is missing something.

Moise2901, 2022-08-02:
Sorry for the delay, I hadn't had access to our setup until today. I did what you suggested and the problem is now gone, the issue was that I wasn't sourcing the correct workspace, thanks for giving me the head's up @zflat ! The issue can be closed now :)

closed 2022-08-02 by Moise2901