RobotWebTools/rosbridge_suite #767: String Escaping for single and double qoute on Noetic

parsha, 2022-07-26:
**Description**
I am using ROSsharp in Unity to communicate with rosbridge through Websocket to send JSON string messages, like "{\"key\" : \"value\"}.
But on the ROS side, when I echo my topic, I recieve data: "{\\u0022key\\u0022 : \\u0022value\\u0022}"
it is interesting that I only have this problem with rosbridge on noetic. I also tested my implementation on melodic and it works without any problem.

- Library Version: Last availble version... 0.11.14-1focal
- ROS Version: noetic
- Platform / OS: ROSsharp running on Unity (Windows), Rosbridge_server running on Ubuntu

parsha, 2022-07-28:
Update: So far I figured out, it is a problem in ROSsharp ... and not a problem from rosbrdige!

closed 2022-08-01 by jtbandes

sebgir, 2023-01-30:
@parsha : I know this issue is not related to RosBridge and has been closed here but did you managed to fix this issue in RosSharp ?

highfield, 2024-08-10:
I just stumbled upon the same problem, but I'm not using RosSharp. I'm using a library I made by myself from scratch. IMHO, the problem is on the ROS-side.

During the inspection of the issue, I also sniffed the network and that's what actually is incoming.

In the **Melodic** release, the `value` field contains a string as the simple serialization of the useful data.
```
{
  ...
  "value": "{\"sprayer\": [], \"barPosition\": 0}",
  ...
}
```

In the **Noetic** release, however, the `value` field looks as a weird progressive serialization of the original structure. Cannot figure out how the string sent is actually composed.

```
0000   00 ff 22 4f cd 39 00 ff 23 4f cd 39 08 00 45 00   .."O.9..#O.9..E.
0010   00 f7 cd 1f 40 00 40 06 55 2e 0a 08 03 2e 0a 08   ....@.@.U.......
0020   00 76 23 82 c5 7d 9a 4e 66 aa 1e 22 05 a3 50 18   .v#..}.Nf.."..P.
0030   01 f5 a3 ee 00 00 81 7e 00 cb 7b 22 6f 70 22 3a   .......~..{"op":
0040   22 73 65 72 76 69 63 65 5f 72 65 73 70 6f 6e 73   "service_respons
0050   65 22 2c 22 73 65 72 76 69 63 65 22 3a 22 5c 2f   e","service":"\/
0060   72 6f 73 61 70 69 5c 2f 67 65 74 5f 70 61 72 61   rosapi\/get_para
0070   6d 22 2c 22 76 61 6c 75 65 73 22 3a 7b 22 76 61   m","values":{"va
0080   6c 75 65 22 3a 22 5c 22 7b 5c 5c 5c 5c 75 30 30   lue":"\"{\\\\u00
0090   32 32 73 70 72 61 79 65 72 5c 5c 5c 5c 75 30 30   22sprayer\\\\u00
00a0   32 32 3a 5b 5d 2c 5c 5c 5c 5c 75 30 30 32 32 62   22:[],\\\\u0022b
00b0   61 72 50 6f 73 69 74 69 6f 6e 5c 5c 5c 5c 75 30   arPosition\\\\u0
00c0   30 32 32 3a 30 7d 5c 22 22 7d 2c 22 72 65 73 75   022:0}\""},"resu
00d0   6c 74 22 3a 74 72 75 65 2c 22 69 64 22 3a 22 39   lt":true,"id":"9
00e0   39 65 39 62 64 64 63 2d 62 63 38 33 2d 34 33 62   9e9bddc-bc83-43b
00f0   63 2d 39 62 64 64 2d 61 39 33 34 64 63 35 37 63   c-9bdd-a934dc57c
0100   31 64 32 22 7d                                    1d2"}
```

Here is some code to mimic the string composition, although that's not exactly what happens.

```
	public record C
	{
		public float[] sprayer { get; set; } = [];
		public int barPosition { get; set; }
	}

	string s1 = JsonSerializer.Serialize(new C());
	Console.WriteLine("s1=" + s1);	//s1={"sprayer":[],"barPosition":0}
	string s2 = JsonSerializer.Serialize(s1);
	Console.WriteLine("s2=" + s2);	//s2="{\u0022sprayer\u0022:[],\u0022barPosition\u0022:0}"
	string s3 = JsonSerializer.Serialize(s2);
	Console.WriteLine("s3=" + s3);	//s3="\u0022{\\u0022sprayer\\u0022:[],\\u0022barPosition\\u0022:0}\u0022"
```

Again, to me, that looks a ROS bridge bug or some breaking change not documented.