RobotWebTools/rosbridge_suite #772: CPU usage increases over time.

obi-t4, 2022-08-03:
**Description**

CPU usage increases over time.

- Library Version:
  installed with apt
  `ros-galactic-rosbridge-suite/focal,now 1.2.0-1focal.20220520.223734 amd64 [installed]`
- ROS Version:
  galactic
- Platform / OS:
  Ubuntu20.04

**Steps To Reproduce**

1. `ros2 run demo_nodes_cpp add_two_ints_server`
   https://github.com/ros2/demos is used.
1. `ros2 run rosbridge_server rosbridge_websocket`
2. Run the following scripts in multiple terminals: (two terminals are OK)
```python3
import roslibpy
import time
import datetime
import sys
from threading import Thread, Lock

ros = roslibpy.Ros(host='localhost', port=9090)
ros.run(timeout=None)

srv = roslibpy.Service(ros, "/add_two_ints", "example_interfaces/srv/AddTwoInts")
srvreq = roslibpy.ServiceRequest({"a": 1, "b": 2})

def srvcb(data):
    print(data)

def srv_errcb(data):
    print(f"error: {data}")

def service():
    while ros.is_connected:
        srv.call(srvreq, callback=srvcb, errback=srv_errcb, timeout=5)
        time.sleep(0.001)

th = Thread(target=service, daemon=True)
th.start()

while True:
    time.sleep(1)
```
1. wait for several hours. 4 or 5 hours.
2. In the beginning, CPU usage is about 45%, but after a few hours, it will exceed 50%, in my environment.

**Expected Behavior**

CPU usage should keep up.

**Actual Behavior**

CPU usage increases slightly over time.

obi-t4, 2022-08-03:
CPU usage increases up to 85%.
[image]

lroop, 2022-09-20:
I also observe this issue, and see it slow down other ROS nodes in my system.

closed 2022-10-31 by achim-k

Mechazo11, 2025-06-20:
Hi @obi-t4 I know this issue was probably closed but recently we saw this issue with a robotics system we are working with where the ROS 2 server is configured to send costmap and other pertinent data to foxglove via rosbridge_websocket. The RAM usage is stable but after around 15 min mark, the RAM usage gradually creeps up until all available memory is used up. Any suggestion on how to go about fixing this issue?