RobotWebTools/rosbridge_suite #775: Exception calling subscribe callback: 'bytearray' object has no attribute 'get'

an99990, 2022-08-11:
**Description**
Sending PointCloud2 message and trying to visualize it in FoxeGlove but keep getting rosbridge error message 

[rosbridge_websocket-1] [ERROR] [1660225884.363841657] [rosbridge_websocket]: Exception calling subscribe callback: 'bytearray' object has no attribute 'get

ROS Version (echo $ROS_DISTRO): Ros2 Galactic
OS Version (grep DISTRIB_CODENAME /etc/lsb-release): Focal

**Steps To Reproduce**
Publish Pointcloud2 msg
launch rosbridge
connect with foxglove

**Expected Behavior**
see pointcloud

**Actual Behavior**
nothing to see

ben-congram, 2023-07-10:
I was receiving a similar error message  when trying to subscribe to a large PointCloud2 msg: `[rosbridge_websocket]: Exception calling subscribe callback: 'bytes' object has no attribute 'get'`. Increasing the value of the `max_message_size` parameter solved the issue for me.

simon-eis, 2024-01-10:
I had a similar error message when trying to subscribe to a video stream. 
[rosbridge_websocket]: Exception calling subscribe callback: 'bytes' object has no attribute 'get'
For me the solution was to downscale the videostream now everything works fine and i can visualize the videostream in foxglove studio. I did not test the fix with increasing max_message_size. The datarate of the original stream was 24.8 MB per image with 20 FPS this computes to 496 MB/s, which was obviously too much for the rosbridge.

closed 2025-02-09 by github-actions[bot]