RobotWebTools/rosbridge_suite #778: Bridge fails to forward data correctly to foxglove on humble

awischnewski, 2022-08-20:
**Description**
There seems to be an issue with the websocket used as an interface to foxglove in the humble package of `rosbridge-suite`. It does not subscribe to topics and therefore does not display the topics in foxglove. 

- Library Version: 1.2.0-1jammy.20220808.155558
- ROS Version: humble
- Platform / OS: docker container based on Ubuntu 22.04 running on Ubuntu 20.04

**Steps To Reproduce**
The setup is distributed among multiple docker containers. 

First container: replays rosbag with LIDAR pointclouds (onto host network). 
Second container: Launches websocket via `ros2 launch rosbridge_server rosbridge_websocket_launch.xml`
foxglove is launched on the host machine. 

**Expected Behavior**
Data should be forwarded to foxglove

**Actual Behavior**
This setup works fine with all containers based on a docker container based on Ubuntu 20.04 and ROS2 galactic. It also works fine if the rosbag container is launched from the 22.04 base with ROS2 humble. It stops working when the websocket container is also launched from the 22.04 / humble.

jtbandes, 2022-08-20:
Can you share the containers?

awischnewski, 2022-08-22:
I have just tested with the publically available images (https://hub.docker.com/_/ros/) `ros:galactic` and `ros:humble` and they show the same behavior. I could share the rosbag privately as well if thats required. 

Not sure at the moment, whether this is an issue with the bridge or rosbag itself as I have experienced another strange topic discovery issue with rosbag on humble today in a context which is unrelated to the bridge.

achim-k, 2022-10-07:
> This kind of points me into the direction that this issue is unrelated to the bridge.

I will close this issue then

closed 2022-10-07 by achim-k