RobotWebTools/rosbridge_suite #783: Multiple webpage refresh crashes Rosbridge server

apoorvcn47, 2022-09-06:
**Description**
I have installed debian package of rosbridge suite for ROS2 Foxy. It runs fine but if i keep refreshing my webpage, it eventually crashes rosbridge server. In the troubleshoot document it mentions that tornado version mismatch can cause a few problems. Where to check which tornado version goes along with Rosbridge 1.2? Since I have used apt-get install to install the package, I don't think I can use rosdep, right?

Following is the Log I get:
[INFO] [launch]: All log files can be found below /home/aw/.ros/log/2022-09-06-15-32-09-003727-ubuntu-75240
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rosbridge_websocket-1]: process started with pid [75242]
[INFO] [rosapi_node-2]: process started with pid [75244]
[rosapi_node-2] 1662492729.234910 [0]    python3: using network interface ens33 (udp/192.168.72.130) selected arbitrarily from: ens33, docker0
[rosbridge_websocket-1] 1662492729.268578 [0]    python3: using network interface ens33 (udp/192.168.72.130) selected arbitrarily from: ens33, docker0
[rosbridge_websocket-1] [INFO] [1662492729.311832492] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9090
[rosbridge_websocket-1] [INFO] [1662492734.211482632] [rosbridge_websocket]: Client connected. 1 clients total.
[rosbridge_websocket-1] [INFO] [1662492734.313060972] [rosbridge_websocket]: [Client 8ca8269e-ef1b-4d75-8cbb-0195a3a5c1ff] Subscribed to /ui/fb/full_markers
[rosbridge_websocket-1] [INFO] [1662492734.314190071] [rosbridge_websocket]: [Client 8ca8269e-ef1b-4d75-8cbb-0195a3a5c1ff] Subscribed to /ui/fb/pallet1_markers
[rosbridge_websocket-1] [INFO] [1662492734.315069162] [rosbridge_websocket]: [Client 8ca8269e-ef1b-4d75-8cbb-0195a3a5c1ff] Subscribed to /ui/fb/pallet2_markers
[rosbridge_websocket-1] [INFO] [1662492734.315881053] [rosbridge_websocket]: [Client 8ca8269e-ef1b-4d75-8cbb-0195a3a5c1ff] Subscribed to /ui/fb/pallet3_markers
[rosbridge_websocket-1] [INFO] [1662492734.316826799] [rosbridge_websocket]: [Client 8ca8269e-ef1b-4d75-8cbb-0195a3a5c1ff] Subscribed to /ui/fb/pallet4_markers
[rosbridge_websocket-1] [INFO] [1662492734.317806328] [rosbridge_websocket]: [Client 8ca8269e-ef1b-4d75-8cbb-0195a3a5c1ff] Subscribed to /ui_sim/fb/failed_part
[rosbridge_websocket-1] [INFO] [1662492734.318707581] [rosbridge_websocket]: [Client 8ca8269e-ef1b-4d75-8cbb-0195a3a5c1ff] Subscribed to /ui_sim/fb/status
[rosbridge_websocket-1] [INFO] [1662492734.323567436] [rosbridge_websocket]: [Client 8ca8269e-ef1b-4d75-8cbb-0195a3a5c1ff] Subscribed to /ui_sim/fb/placeable_list
[rosbridge_websocket-1] [INFO] [1662492734.325108644] [rosbridge_websocket]: [Client 8ca8269e-ef1b-4d75-8cbb-0195a3a5c1ff] Subscribed to /ui/fb/robot_connected
[rosbridge_websocket-1] [INFO] [1662492736.087180626] [rosbridge_websocket]: Client disconnected. 0 clients total.
[rosbridge_websocket-1] [INFO] [1662492736.189102910] [rosbridge_websocket]: Client connected. 1 clients total.
[rosbridge_websocket-1] [INFO] [1662492736.252766050] [rosbridge_websocket]: [Client d09319ee-1fe5-4e7f-a24c-455b6b9fcc3b] Subscribed to /ui/fb/full_markers
[rosbridge_websocket-1] [INFO] [1662492736.254386085] [rosbridge_websocket]: [Client d09319ee-1fe5-4e7f-a24c-455b6b9fcc3b] Subscribed to /ui/fb/pallet1_markers
[rosbridge_websocket-1] [INFO] [1662492736.255672828] [rosbridge_websocket]: [Client d09319ee-1fe5-4e7f-a24c-455b6b9fcc3b] Subscribed to /ui/fb/pallet2_markers
[rosbridge_websocket-1] [INFO] [1662492736.257348837] [rosbridge_websocket]: [Client d09319ee-1fe5-4e7f-a24c-455b6b9fcc3b] Subscribed to /ui/fb/pallet3_markers
[rosbridge_websocket-1] [INFO] [1662492736.259179276] [rosbridge_websocket]: [Client d09319ee-1fe5-4e7f-a24c-455b6b9fcc3b] Subscribed to /ui/fb/pallet4_markers
[rosbridge_websocket-1] [INFO] [1662492736.260326568] [rosbridge_websocket]: [Client d09319ee-1fe5-4e7f-a24c-455b6b9fcc3b] Subscribed to /ui/fb/robot_connected
[rosbridge_websocket-1] [INFO] [1662492736.946170828] [rosbridge_websocket]: Client disconnected. 0 clients total.
[rosbridge_websocket-1] [INFO] [1662492737.039601475] [rosbridge_websocket]: Client connected. 1 clients total.
[rosbridge_websocket-1] [INFO] [1662492737.079858959] [rosbridge_websocket]: [Client 81247b43-8c75-4b92-8e74-522d7dcbf712] Subscribed to /ui/fb/full_markers
[rosbridge_websocket-1] [INFO] [1662492737.080645017] [rosbridge_websocket]: [Client 81247b43-8c75-4b92-8e74-522d7dcbf712] Subscribed to /ui/fb/pallet1_markers
[rosbridge_websocket-1] [INFO] [1662492737.081891936] [rosbridge_websocket]: [Client 81247b43-8c75-4b92-8e74-522d7dcbf712] Subscribed to /ui/fb/pallet2_markers
[rosbridge_websocket-1] [INFO] [1662492737.086566073] [rosbridge_websocket]: [Client 81247b43-8c75-4b92-8e74-522d7dcbf712] Subscribed to /ui/fb/pallet3_markers
[rosbridge_websocket-1] [INFO] [1662492737.088225882] [rosbridge_websocket]: [Client 81247b43-8c75-4b92-8e74-522d7dcbf712] Subscribed to /ui/fb/pallet4_markers
[rosbridge_websocket-1] [INFO] [1662492737.089469424] [rosbridge_websocket]: [Client 81247b43-8c75-4b92-8e74-522d7dcbf712] Subscribed to /ui/fb/robot_connected
[rosbridge_websocket-1] [INFO] [1662492737.549564256] [rosbridge_websocket]: Client disconnected. 0 clients total.
[rosbridge_websocket-1] [INFO] [1662492737.651192320] [rosbridge_websocket]: Client connected. 1 clients total.
[rosbridge_websocket-1] [INFO] [1662492737.701295198] [rosbridge_websocket]: [Client 2bd74b31-5861-4085-af86-0ef84579a2d3] Subscribed to /ui/fb/full_markers
[rosbridge_websocket-1] [INFO] [1662492737.702374283] [rosbridge_websocket]: [Client 2bd74b31-5861-4085-af86-0ef84579a2d3] Subscribed to /ui/fb/pallet1_markers
[rosbridge_websocket-1] [INFO] [1662492737.703679360] [rosbridge_websocket]: [Client 2bd74b31-5861-4085-af86-0ef84579a2d3] Subscribed to /ui/fb/pallet2_markers
[rosbridge_websocket-1] [INFO] [1662492737.705692660] [rosbridge_websocket]: [Client 2bd74b31-5861-4085-af86-0ef84579a2d3] Subscribed to /ui/fb/pallet3_markers
[rosbridge_websocket-1] [INFO] [1662492737.707388676] [rosbridge_websocket]: [Client 2bd74b31-5861-4085-af86-0ef84579a2d3] Subscribed to /ui/fb/pallet4_markers
[rosbridge_websocket-1] [INFO] [1662492737.708900319] [rosbridge_websocket]: [Client 2bd74b31-5861-4085-af86-0ef84579a2d3] Subscribed to /ui/fb/robot_connected
[rosbridge_websocket-1] [INFO] [1662492738.064621269] [rosbridge_websocket]: Client disconnected. 0 clients total.
[ERROR] [rosbridge_websocket-1]: process has died [pid 75242, exit code -11, cmd '/opt/ros/foxy/lib/rosbridge_server/rosbridge_websocket --ros-args --params-file /tmp/launch_params_vqse4uzf --params-file /tmp/launch_params_9leetaiu --params-file /tmp/launch_params_jz9gye9r --params-file /tmp/launch_params_4i6_684g --params-file /tmp/launch_params_7ztmwddg --params-file /tmp/launch_params_3us2okw4 --params-file /tmp/launch_params_45p84il2'].

- Library Version: 1.2.0
- ROS Version: Foxy
- Platform / OS: Ubuntu 20.04

zflat, 2022-09-09:
@apoorvcn47 Could you try the branch in https://github.com/RobotWebTools/rosbridge_suite/pull/741? When you refresh it may be the same issue I have seen in https://github.com/RobotWebTools/rosbridge_suite/issues/744.

apoorvcn47, 2022-09-10:
@zflat This branch is definitely more stable. I have to refresh my webpage really fast to make it crash. It does crash if refreshed fairly quickly numerous times but the crashing frequency is way lower than what it was before.

zflat, 2022-09-22:
Hi @apoorvcn47 It is unfortunate that you are still seeing the crash. I wonder if there are any other additional issues. The log that you posted does not have a stack trace. Maybe you could try to run the bridge manually using `ros2 run ...` instead of from a launch? Then when it crashes see if you get a stacktrace in the terminal?

PR #803 (2022-10-17): Improve robustness for multiple client connections
**Public API Changes**
None

**Description**
Supersedes #741
Fixes #783
Fixes #758

This PR improves multi-client connection robustness by 
- avoiding unregistering of subscribers if one client disconnects while all others are still waiting for the [initial callback](https://github.com/RobotWebTools/rosbridge_suite/blob/87b3bf0a1a69a9bcfad5f32cb75303881607a5f1/rosbridge_library/src/rosbridge_library/internal/subscribers.py#L222-L238)
- avoiding a deadlock that could occur when new clients connect, subscribe to a topic and then immediately disc [... cut]
closed 2022-10-17 by achim-k