RobotWebTools/rosbridge_suite #784: double array

AltinayGrass, 2022-09-06:
Hello,

I am getting below error but in rqt everything looks fine. Is it bug ?

`[ERROR] [1662500907.234621770] [rosbridge_websocket]: [Client d6d0ddef-ca19-4008-a300-77fc29d3a694] [id: publish:/servo_node/delta_joint_cmds:607] publish: The 'velocities' field must be a set or sequence and each value of type 'float' and each double in [-179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000, 179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000]`

AltinayGrass, 2022-09-07:
Sorry for the false alarm. I checked servo_node source. The alarm is not only specific for double values. It is also combination of some missing parameters.

achim-k, 2022-10-07:
@AltinayGrass looks like this issue was due to wrong usage of the publish operation (incorrect payload?). Will close this issue, feel free to re-open.

closed 2022-10-07 by achim-k