RobotWebTools/rosbridge_suite #790: Calling the /rosapi/topics services on galactic crashes the bridge

swhart115, 2022-09-26:
**Description**
Hi, I am trying to call the **/rosapi/topics** service from my javascript, and whenever i do, the rosbridge crashes.  

- ROS Version: galactic
- Platform / OS: Ubuntu 20.04

**Steps To Reproduce**
Here is my Java:

> var topicsClient = new ROSLIB.Service({
>            ros : self.ros,
>            name : '/rosapi/topics',
>            serviceType : 'rosapi/Topics'
>         });
> 
> console.log("calling service: /rosapi/topics"); 
> var request = new ROSLIB.ServiceRequest();
> topicsClient.callService(request, function(result) {        
>     console.log("got " + result.topics.length + " topics");
> });

**Actual Behavior**
Note, i seem to get the response back, as i see this in the console output:
> calling service: /rosapi/topics
> got 36 topics

but i get the following error on the command line and have to restart the bridge:

> 
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [rosbridge_websocket-1]: process started with pid [3819303]
> [INFO] [rosapi_node-2]: process started with pid [3819305]
> [rosbridge_websocket-1] [INFO] [1664228069.439557694] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9090
> [rosbridge_websocket-1] [INFO] [1664228072.486532385] [rosbridge_websocket]: Client connected. 1 clients total.
> [rosbridge_websocket-1] [INFO] [1664228072.498579467] [rosbridge_websocket]: Client connected. 2 clients total.
> [rosapi_node-2] Traceback (most recent call last):
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/rosapi/rosapi_node", line 330, in <module>
> [rosapi_node-2]     main()
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/rosapi/rosapi_node", line 321, in main
> [rosapi_node-2]     rclpy.spin(node)
> [rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/__init__.py", line 196, in spin
> [rosapi_node-2]     executor.spin_once()
> [rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 712, in spin_once
> [rosapi_node-2]     raise handler.exception()
> [rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
> [rosapi_node-2]     self._handler.send(None)
> [rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 418, in handler
> [rosapi_node-2]     await call_coroutine(entity, arg)
> [rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 372, in _execute_service
> [rosapi_node-2]     response = await await_or_execute(srv.callback, request, srv.srv_type.Response())
> [rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 107, in await_or_execute
> [rosapi_node-2]     return callback(*args)
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/rosapi/rosapi_node", line 228, in get_message_details
> [rosapi_node-2]     dict_to_typedef(d) for d in objectutils.get_typedef_recursive(request.type)
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 102, in get_typedef_recursive
> [rosapi_node-2]     return _get_typedefs_recursive(type, [])
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 229, in _get_typedefs_recursive
> [rosapi_node-2]     return _get_subtypedefs_recursive(typedef, typesseen)
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 239, in _get_subtypedefs_recursive
> [rosapi_node-2]     typedefs = typedefs + _get_typedefs_recursive(fieldtype, typesseen)
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 229, in _get_typedefs_recursive
> [rosapi_node-2]     return _get_subtypedefs_recursive(typedef, typesseen)
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 239, in _get_subtypedefs_recursive
> [rosapi_node-2]     typedefs = typedefs + _get_typedefs_recursive(fieldtype, typesseen)
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 227, in _get_typedefs_recursive
> [rosapi_node-2]     typedef = get_typedef(type)
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 82, in get_typedef
> [rosapi_node-2]     return _get_typedef(instance)
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 168, in _get_typedef
> [rosapi_node-2]     fieldtypes.append(_type_name(field_type, field_instance))
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 257, in _type_name
> [rosapi_node-2]     return _type_name_from_instance(instance)
> [rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 261, in _type_name_from_instance
> [rosapi_node-2]     mod = instance.__module__
> [rosapi_node-2] AttributeError: 'array.array' object has no attribute '__module__'
> [ERROR] [rosapi_node-2]: process has died [pid 3819305, exit code 1, cmd '/home/swhart/ros2/brash/install/rosapi/lib/rosapi/rosapi_node --ros-args -r __node:=rosapi --params-file /tmp/launch_params_ltk5ftnz --params-file /tmp/launch_params_ntcfoh0i --params-file /tmp/launch_params_wmod3urr'].

any help will be much appreciated. 
Thanks!

achim-k, 2022-10-07:
I could not reproduce this on `humble` with the simple javascript snippet below. 

```node
const roslib = require("roslib")
const ros = new roslib.Ros({ url: 'ws://127.0.0.1:9090' })
const topicsClient = new roslib.Service({ ros: ros, name: '/rosapi/topics' })
topicsClient.callService(new roslib.ServiceRequest(), console.log)
```

returns
```
ServiceResponse {
  topics: [
    '/client_count',
    '/connected_clients',
    '/parameter_events',
    '/rosout'
  ],
  types: [
    'std_msgs/msg/Int32',
    'rosbridge_msgs/msg/ConnectedClients',
    'rcl_interfaces/msg/ParameterEvent',
    'rcl_interfaces/msg/Log'
  ]
}
```

Could you try that snippet yourself? Do you have some weird topic names that might cause that error?

swhart115, 2022-10-11:
Thanks @achim-k for your response.  I tried the above on my galactic setup, but still no luck.  I think it was basically the same as what i had been running, though i had been setting the ros object as a class variable.

In any case, i did some more digging based on your suggestion that it might be something to do with my specific topics/msgs.  This helped, as the culprit seems to be a custom message i have that is using some `uint8[]` fields. My message def is:  
```
uint8[] stream_id  
uint8[] sequence
uint8[] length
```
As a test i tried changing these to other primitive types, just to see, and regardless of what type the arrays were, the rosbridge spits out the reported error when the topic service is called.   Non-array fields of the same types seem fine though.  I also tried changing the field names, but that didnt seem to make any difference.

achim-k, 2022-10-11:
It would be good if you could provide detailed instructions on how to reproduce this issue. I assume, that you already see the error when calling `ros2 service call /rosapi/topics rosapi_msgs/srv/Topics`?

Can you figure out which topic is responsible for the crash? Could you provide the message files?

swhart115, 2022-10-11:
I can try to extract runnable code chunks to reproduce, but here is how to set it up:

1. Define a message with one or more `uint8[]` fields, like my example above. 
2. Create and run a ROS2 node that publishes or subscribes to this message type. (My original code has a subscriber to the message type in question, but nothing is being published on the topic so I don't think it matters. I've just created the subscriber.) 
3. start the rosbridge server
4. use the roslib javascript library to call the `/rosapi/topics` service. The rosbridge server should spit out the error indicated in the original post above.

note that just calling the service from the command line (or another node), as you suggest does not result in any issues. it is only when the service is called from the javascript. 

Also, to reiterate, i am using galactic on a standard 20.04 Ubuntu release.

swhart115, 2022-10-12:
Okay, i wrote a very simple python node that sets up a publisher for an `Int8MultiArray` (it doesnt even actually publish anything on it).  i picked that cause it has an array of primitives in it, which is similar to what i was seeing with my custom packages and applications.

Here is my node code:
```
import rclpy
from rclpy.node import Node
from std_msgs.msg import Int8MultiArray

class MsgPubTest(Node):
    def __init__(self):
        super().__init__('msg_pub_test')        
        self._pub = self.create_publisher(Int8MultiArray, "/int_array_pub", 1)

def main(args=None):
    rclpy.init(args=args)
    t = MsgPubTest()
    rclpy.spin(t)
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

I then run the rosbridge:
`ros2 launch rosbridge_server rosbridge_websocket_launch.xml`

and my javascript application, that calls the `/rosapi/topics` service:
```
const ROSLIB = require('roslib')
const ros = new ROSLIB.Ros({ url: 'ws://localhost:9090' })
const topicsClient = new ROSLIB.Service({ ros: ros, name: '/rosapi/topics' })
topicsClient.callService(new ROSLIB.ServiceRequest(), console.log)
```

While the service seems to return appropriately (i see the correct response in the browser console window), the bridge spits out this error:

```
[INFO] [rosapi_node-2]: process started with pid [1318554]
[rosbridge_websocket-1] [INFO] [1665602272.558512264] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9090
[rosbridge_websocket-1] [INFO] [1665602276.273130941] [rosbridge_websocket]: Client connected. 1 clients total.
[rosbridge_websocket-1] [INFO] [1665602276.353760743] [rosbridge_websocket]: Client connected. 2 clients total.
[rosapi_node-2] Traceback (most recent call last):
[rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/rosapi/rosapi_node", line 330, in <module>
[rosapi_node-2]     main()
[rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/rosapi/rosapi_node", line 321, in main
[rosapi_node-2]     rclpy.spin(node)
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/__init__.py", line 196, in spin
[rosapi_node-2]     executor.spin_once()
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 712, in spin_once
[rosapi_node-2]     raise handler.exception()
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
[rosapi_node-2]     self._handler.send(None)
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 418, in handler
[rosapi_node-2]     await call_coroutine(entity, arg)
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 372, in _execute_service
[rosapi_node-2]     response = await await_or_execute(srv.callback, request, srv.srv_type.Response())
[rosapi_node-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 107, in await_or_execute
[rosapi_node-2]     return callback(*args)
[rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/rosapi/rosapi_node", line 228, in get_message_details
[rosapi_node-2]     dict_to_typedef(d) for d in objectutils.get_typedef_recursive(request.type)
[rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 102, in get_typedef_recursive
[rosapi_node-2]     return _get_typedefs_recursive(type, [])
[rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 227, in _get_typedefs_recursive
[rosapi_node-2]     typedef = get_typedef(type)
[rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 82, in get_typedef
[rosapi_node-2]     return _get_typedef(instance)
[rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 168, in _get_typedef
[rosapi_node-2]     fieldtypes.append(_type_name(field_type, field_instance))
[rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 257, in _type_name
[rosapi_node-2]     return _type_name_from_instance(instance)
[rosapi_node-2]   File "/home/swhart/ros2/brash/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 261, in _type_name_from_instance
[rosapi_node-2]     mod = instance.__module__
[rosapi_node-2] AttributeError: 'array.array' object has no attribute '__module__'
[ERROR] [rosapi_node-2]: process has died [pid 1318554, exit code 1, cmd '/home/swhart/ros2/brash/install/rosapi/lib/rosapi/rosapi_node --ros-args -r __node:=rosapi --params-file /tmp/launch_params_5y428pzo --params-file /tmp/launch_params_0dtu63hx --params-file /tmp/launch_params_alj71qe2'].

```

achim-k, 2022-10-21:
I failed to reproduce your issue. Which rosbridge version / commit are you using?

closed 2023-11-21 by github-actions[bot]