RobotWebTools/rosbridge_suite #791: InvalidServiceException: Service /rosapi/topics_and_raw_types does not exist (Docker + Fast-DDS Discovery Server)

kledom, 2022-09-28:
**Description**

I'm running ROS2 humble using [Fast-DDS discovery Server](https://docs.ros.org/en/humble/Tutorials/Advanced/Discovery-Server/Discovery-Server.html). This is necessary because I want to deploy ROS over a VPN that does not support multicast. To make things even more complicated, I would also like to use docker-compose.

I have a docker-compose file as a template that I am absolutely sure of that it works with basic ROS2 nodes (e.g.  demo_nodes_cpp talker and listener). In fact, I've already deployed it several times in similar scenarios. However, using it with rosbridge_suite leads to errors.

- Library Version: 1.3.0
- ROS Version: humble
- Platform / OS: Ubuntu 22.04 (Host), ros:humble-ros-core (Container)
- rosbridge_suite client: e.g. foxglove-studio.

**Steps To Reproduce**

Dockerfile
```
ARG ROS_VERSION=humble
FROM ros:${ROS_VERSION}-ros-core

ARG ROS_VERSION

# Install rosbridge_suite
RUN apt-get update -qq && \
    apt-get install -y ros-${ROS_VERSION}-rosbridge-server && \
    rm -rf /var/lib/apt/lists/*

CMD ["ros2", "launch", "rosbridge_server", "rosbridge_websocket_launch.xml"]
```

docker-compose.yml
```
version: "3.3"

services:
  discovery:
    image: ros:humble-ros-core
    command: fastdds discovery -i 0 -p 11811
    network_mode: host

  bridge_suite:
    depends_on:
      - discovery
    build:
      context: .
      dockerfile: ./Dockerfile
    environment:
      - RMW_IMPLEMENTATION=rmw_fastrtps_cpp
      - FASTRTPS_DEFAULT_PROFILES_FILE=/dds_config.xml
      - ROS_DISCOVERY_SERVER=127.0.0.1:11811
      - PYTHONUNBUFFERED=1
    volumes:
      - ${PWD}/fastRTPS_no_SHM.xml:/dds_config.xml
    network_mode: host
```

fastRTPS_no_SHM.xml

- (disable shared memory transport, currently required, I don't like it either)

```
<?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <transport_descriptors>
            <transport_descriptor>
                <transport_id>udp_transport</transport_id>
                <type>UDPv4</type>
            </transport_descriptor>
        </transport_descriptors>
        <participant profile_name="participant_profile" is_default_profile="true">
            <rtps>
                <userTransports>
                    <transport_id>udp_transport</transport_id>
                </userTransports>
                <useBuiltinTransports>false</useBuiltinTransports>
            </rtps>
        </participant>
    </profiles>
</dds>
```

**Expected Behavior**

Client connects to rosbridge_suite successfully.

**Actual Behavior**

Starting rosbridge_suite with `ros2 launch rosbridge_server rosbridge_websocket_launch.xml` does not produce any noticeable output. However, as soon as a clients connects, the following error message gets printed repeatedly:

> [Client 2f57dba0-9163-4c9d-9ee2-eb9a814351ed] [id: call_service:/rosapi/topics_and_raw_types:4] call_service InvalidServiceException: Service /rosapi/topics_and_raw_types does not exist

Without the environment variable ROS_DISCOVERY_SERVER everything works as expected. But of course the Discovery Server is not used then.

kledom, 2022-09-29:
Here is another, simpler way to reproduce the error.

**Steps To Reproduce**

Run the following lines in 4 different terminals:

Terminal 1
```bash
$ fastdds discovery -i 0 -p 11811
```

```
### Server is running ###
  Participant Type:   SERVER
  Server ID:          0
  Server GUID prefix: 44.53.00.5f.45.50.52.4f.53.49.4d.41
  Server Addresses:   UDPv4:[0.0.0.0]:1181
```

Terminal 2
```bash
$ export ROS_DISCOVERY_SERVER=127.0.0.1:11811
$ ros2 run demo_nodes_cpp talker 
```

```
[INFO] [1664448664.814713110] [talker]: Publishing: 'Hello World: 1'
[INFO] [1664448665.896628676] [talker]: Publishing: 'Hello World: 2'
[INFO] [1664448666.979044302] [talker]: Publishing: 'Hello World: 3'
...
```
Terminal 3
```bash
$ export ROS_DISCOVERY_SERVER=127.0.0.1:11811
$ ros2 run demo_nodes_cpp listener
```

```
[INFO] [1664448676.726041410] [listener]: I heard: [Hello World: 12]
[INFO] [1664448677.808021970] [listener]: I heard: [Hello World: 13]
[INFO] [1664448678.890804089] [listener]: I heard: [Hello World: 14]
...
```

Terminal 4
```bash
$ export ROS_DISCOVERY_SERVER=127.0.0.1:11811
$ ros2 launch rosbridge_server rosbridge_websocket_launch.xml
```

```
[INFO] [launch]: All log files can be found below ...
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rosbridge_websocket-1]: process started with pid [55133]
[INFO] [rosapi_node-2]: process started with pid [55135]
[rosbridge_websocket-1] [INFO] [1664449489.660900232] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9090
```

Connect a rosbridge client (e.g. (Foxglove Studio)[https://foxglove.dev/].

**Result:**
```
[INFO] [1664449490.937003291] [rosbridge_websocket]: Client connected. 1 clients total.
[rosbridge_websocket-1] [ERROR] [1664449490.939020418] [rosbridge_websocket]: [Client 875fd850-fd56-464b-b376-5df102ee2ecc] [id: call_service:/rosapi/topics_and_raw_types:1] call_service InvalidServiceException: Service /rosapi/topics_and_raw_types does not exist
[rosbridge_websocket-1] [ERROR] [1664449493.976456874] [rosbridge_websocket]: [Client 875fd850-fd56-464b-b376-5df102ee2ecc] [id: call_service:/rosapi/topics_and_raw_types:2] call_service InvalidServiceException: Service /rosapi/topics_and_raw_types does not exist
```

The problem is probably that rosapi services are missing from `node_handle.get_service_names_and_types()`.
https://github.com/RobotWebTools/rosbridge_suite/blob/c624f97f0b7ea8ee64386790b0ed99d81d01fbee/rosbridge_library/src/rosbridge_library/internal/services.py#L108

This is the returned dict:

> {'/rosbridge_websocket/describe_parameters': ['rcl_interfaces/srv/DescribeParameters'], '/rosbridge_websocket/get_parameter_types': ['rcl_interfaces/srv/GetParameterTypes'], '/rosbridge_websocket/get_parameters': ['rcl_interfaces/srv/GetParameters'], '/rosbridge_websocket/list_parameters': ['rcl_interfaces/srv/ListParameters'], '/rosbridge_websocket/set_parameters': ['rcl_interfaces/srv/SetParameters'], '/rosbridge_websocket/set_parameters_atomically': ['rcl_interfaces/srv/SetParametersAtomically']}

achim-k, 2022-10-07:
This is not a rosbridge issue but a ros2 node / service discovery issue. https://github.com/ros2/rmw_fastrtps/issues/639 seems the correct place to file this issue, I will close this one.

closed 2022-10-07 by achim-k

kledom, 2022-11-17:
For all others who have this problem: To use the rosbridge_uite with the Fast DDS Discovery Server, the SUPER_CLIENT discovery protocol must be activated in the DDS config file before `ros2 launch rosbridge_server rosbridge_websocket_launch.xml`:

```
<?xml version="1.0" encoding="UTF-8"?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <participant profile_name="super_client_profile" is_default_profile="true">
        <rtps>
            <builtin>
                <discovery_config>
                    <discoveryProtocol>SUPER_CLIENT</discoveryProtocol>
                    <discoveryServersList>
                        <RemoteServer prefix="44.53.00.5f.45.50.52.4f.53.49.4d.41">
                            <metatrafficUnicastLocatorList>
                                <locator>
                                    <udpv4>
                                        <address>127.0.0.1</address>
                                        <port>11811</port>
                                    </udpv4>
                                </locator>
                            </metatrafficUnicastLocatorList>
                        </RemoteServer>
                    </discoveryServersList>
                </discovery_config>
            </builtin>
        </rtps>
    </participant>
</profiles>
```

@achim-k  I don't know if this is relevant enough to include in the documentation?

achim-k, 2022-11-18:
> @achim-k I don't know if this is relevant enough to include in the documentation?

I don't think so, it seems very specific. But thanks for posting the solution here, might help others with the same problem.

groisne, 2023-06-22:
Hello, I am also having this '/rosapi/tpics_and_raw_types' error message when I use foxglove to connect to ros1-indigo remote machine. I set up ROS_MASTER_URI and ROS_HOSTNAME according to http://wiki.ros.org/ROS/NetworkSetup, but I still can't get the topics. One thing I found is that https://github.com/RobotWebTools/rosbridge_suite/tree/ros1/rosapi/srv has the TopicsAndRawTypes.srv file but not for the indigo branch which I am using. I tried to add manually this file to my remote machine and catkin_make again my workspace but it is still not working. Do you have any ideas ?