RobotWebTools/rosbridge_suite #810: Adding namespace in the launch file is not working as expected

hardesh-peppermint, 2022-10-13:
**Description**
Adding a namespace does not work with rosapi node. 

- Library Version: Latest
- ROS Version: Humble
- Platform / OS: Ubuntu 22

**Steps To Reproduce**
Add a namespace while launching the rosapi node using 

**Expected Behavior**
Everything including nodes, topics, services should be inside that namespace. 

**Actual Behavior**
At the moment, only the nodes and some of the services are going inside the namespace. The `rosapi` topics are services are node going inside the namespace. https://github.dev/RobotWebTools/rosbridge_suite/blob/413f844a27281bc02685b600c1756ffecd75a4e8/rosapi/scripts/rosapi_node#L101 in this file they've been written with global namespace. Not sure if directly removing the namespace will solve this issue as it might break other things. 

Sharing my modified launch file with namespace: 
[rosbridge_websocket_launch.xml.txt](https://github.com/RobotWebTools/rosbridge_suite/files/9776967/rosbridge_websocket_launch.xml.txt)

[image]

Sharing some other screenshots:
[image]

achim-k, 2022-10-17:
You are right, it should be possible to launch the rosapi / rosbridge_server nodes in a different namespace. The problem is however, that many websocket clients assume that the `rosapi` node is under the global namespace. For roslibjs this is the case in the following places: https://github.com/RobotWebTools/roslibjs/search?q=%2Frosapi

Feel free to submit a PR that removes the leading `/` for all services created [here](https://github.com/RobotWebTools/rosbridge_suite/blob/df385330dd60d99ff2875eb6c5ac0beda6290627/rosapi/scripts/rosapi_node#L91-L130) and a corresponding PR on the [roslibjs](https://github.com/RobotWebTools/roslibjs) repo. Even with a custom namespace, everything should still work since `rosapi` and `rosbridge_server` are launched in the same namespace.

hardesh-peppermint, 2022-10-18:
Okay, thanks. I'll open up a PR. Will this be merged into humble branch?

hardesh-peppermint, 2022-10-18:
Hi, 
I've opened a new PR. Is it necessary to open a new PR in roslibjs? 
If the user is not using any namespace while launching the nodes, then by default the nodes should go into global namespace. So it won't be an issue I think

hardesh-peppermint, 2022-10-18:
Hi @achim-k ,
The default launch file provided in the rosbridge_server package does not have a namespace argument. So I think there won't be a direct breaking change for regular users of this library, as the topics will directly go into the global namespace. 

I just checked by launching my modified launch file to see if the rosapi services are going into different namespace which is happening as expected. Unfortunately, I've not tested those individual services in rosapi. I just checked if there are any existing testcases in that package. But there aren't any relevant testcases. So I'll have to think about how to write a good unitest.

As I'm writing this, I've realised that I don't really need rosapi. I'm directly using rosbridge_server and afaik it's not dependent on rosapi. I'm using rosbridge_server with a dart based client. I've not used roslibjs in my application

zaher88abd, 2022-11-03:
Hello @hardesh-peppermint 
I just read your are using Rosbridge_server with Dart, I believe it's a Flutter app. Please, may you explain more about your project? like how you did the communication between Ros2 and Flutter by Rosbridge?. 
I have idea about project for connect robot with Futter app.

hardesh-peppermint, 2022-11-04:
Hi @zaher88abd,
I've just created some websocket connections in my flutter app that directly connects with the rosbridge server. The protocol is mentioned in the design docs of this repo. You can also refer to the implementation of websocket client protocols in roslibjs or the rust client. I had referred to the rust client library

zaher88abd, 2022-11-04:
Thanks @hardesh-peppermint

closed 2023-12-06 by github-actions[bot]