RobotWebTools/rosbridge_suite #811: An argument is missing in the 'outgoing' function of the class Protocol.

ribes4, 2022-10-14:
**Description**
In ROS1 in version 0.11.15 a bug has been introduced. I was working with rosbridge via TCP and I got the following traceback in the terminal:

    Traceback (most recent call last):
      File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
        self.run()
      File "/opt/ros/noetic/lib/python3/dist-packages/rosbridge_library/internal/services.py", line 78, in run
        self.error(e)
      File "/opt/ros/noetic/lib/python3/dist-packages/rosbridge_library/capabilities/call_service.py", line 116, in _failure
        self.protocol.send(outgoing_message)
      File "/opt/ros/noetic/lib/python3/dist-packages/rosbridge_library/protocol.py", line 277, in send
        self.outgoing(serialized, compression)
    TypeError: send_message() takes from 1 to 2 positional arguments but 3 were given

Looking at the code I have seen that the `outgoing()` function working  with websocket accepts a new parameter named `compression`. But in TCP it doesn't work because the `outgoing` function is not overridden. Consequently the error shown above appears, because the base function does not accept that argument.

- Library Version: 0.11.15
- ROS Version: Noetic
- Platform / OS: Ubuntu 20.04

**Steps To Reproduce**
Just establish a TCP connection and start transmission.

achim-k, 2022-10-17:
@ribes4 could you confirm that #812 fixes the issue? I tested it locally but it would be great if you could test it as well. I didn't add a unit test as it seemed to much work and on ros2 rosbridge_server does not provide UDP/TCP support anymore.

Will create a new release, which should land in noetic with the next sync [expected to be early november](https://discourse.ros.org/t/noetic-sync-before-roscon-iros/27646/4).

ribes4, 2022-10-18:
@achim-k Tested and works correctly in a TCP transmission. Thanks for the quick response.

achim-k, 2022-10-18:
Thanks for confirming!

closed 2022-10-18 by achim-k

weifeng-lt, 2022-11-18:
The function `outgoing` in line 299 still has only one argument while line 269/271/277 still call `outgoing` with two arguments. The bug has not been fixed !

achim-k, 2022-11-18:
@weifeng-lt could you provide some [permanent links](https://docs.github.com/en/repositories/working-with-files/using-files/getting-permanent-links-to-files) to the mentioned lines of code?

weifeng-lt, 2022-12-26:
https://github.com/RobotWebTools/rosbridge_suite/blob/b0d0d711931d8ab7a5f079f948ced8875b4dc5d9/rosbridge_library/src/rosbridge_library/protocol.py

achim-k, 2023-01-04:
> https://github.com/RobotWebTools/rosbridge_suite/blob/b0d0d711931d8ab7a5f079f948ced8875b4dc5d9/rosbridge_library/src/rosbridge_library/protocol.py

You are right, that the `outgoing` function does not specify the additional `compression` argument. But on the other hand, this function is supposed to be overwritten, so it is never called. Could you elaborate how you are using rosbridge_server and why this function is called in your case? Nevertheless, we should still add the missing argument definition here

https://github.com/RobotWebTools/rosbridge_suite/blob/b0d0d711931d8ab7a5f079f948ced8875b4dc5d9/rosbridge_library/src/rosbridge_library/protocol.py#L229-L237

PR #812 (2022-10-17): Fix send_message being called with wrong arguments.
**Public API Changes**
None

**Description**
Fixes #811, a regression on the latest release when using TCP/UDP transport