RobotWebTools/rosbridge_suite #818: Unable To Import Custom Msg From Custom Package

CrawlerWheel, 2022-10-19:
**Description**
Hi, Authors.I define a custom message. As shown in the figure, I can get the message structure in ros, after modify the env var.
[image]
But,when I use distant js program in Windows to subscribe rostopic `opencv_test/TargetArray` via rosbridge in Ubuntu, rosbridge alert:
[image]
It seems like that rosbridge can locate my custom package in Ubuntu. But, why its goal is to find  `opencv_test.msg`? Actually，this right goal is to find `TargetArray.msg`, just like ROS do.This is confusing to me！

- Library Version:ros-melodic-rosbridge-library/bionic,now 0.11.14-1bionic.20220615.170004 arm64
- ROS Version: Melodic
- Platform / OS: Ubuntu18.04

**Steps To Reproduce**
js code:
```js
const ROSLIB = require('roslib')
const eventemitter2 = require('eventemitter2')

// Connecting to ROS
var ros = new ROSLIB.Ros({
  url : 'ws://192.168.0.40:9090'
});

// If there is an error on the backend, an 'error' emit will be emitted.
ros.on('error', function(error) {
  console.log(error);
});

// Find out exactly when we made a connection.
ros.on('connection', function() {
  console.log('Connection made!');
});

ros.on('close', function() {
  console.log('Connection closed.');

});

// Create a connection to the rosbridge WebSocket server.
ros.connect('ws://localhost:9090');

var listener_mark = new ROSLIB.Topic({
  ros : ros,
  name : '/detect/MarkInfo',
  messageType : 'opencv_test/TargetArray'
});

listener_mark.subscribe(function(message) {
  console.log("gei it")
  //console.log('Received message on ' + listener.name +  "time: "+JSON.stringify(message.header.stamp));
  // If desired, we can unsubscribe from the topic as well.
  //listener.unsubscribe();
});
```
and output at terminal in js program :
[image]

**Expected Behavior**
I wana it can let distant js program subscribe rostopic `opencv_test/TargetArray` ，and parse  `opencv_test/TargetArray` message successfully.

**Actual Behavior**
distant js program can`t get anything.

achim-k, 2022-10-19:
Is the ROS environment loaded (usually via `source devel/setup.bash`) in the terminal in which you launch `rosbridge_server`?

If it's loaded, can you confirm that the following succeeds:
```bash
rosmsg show opencv_test/TargetArray
python -c 'from opencv_test.msg import TargetArray'
```

CrawlerWheel, 2022-10-20:
Yes，I loaded, I have added `export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/wheeltec/vins_ws/src` to .bashrc. And command `rosmsg show opencv_test/TargetArray` can return the message structure successfully.
[image]
[image]

achim-k, 2022-10-21:
OK good to know it's working now. Closing this issue then.

closed 2022-10-21 by achim-k