RobotWebTools/rosbridge_suite #829: Subscriptions stop working after longer periods of time

lkeselman, 2022-12-16:
**Description**
Periodically after relatively long periods of time, my Javascript client webpage no longer receives messages from subscriptions. 

- I can echo the messages using `rostopic echo` so they are in fact published.
- When refreshing the webpage, the rosbridge logs indicate a successful subscription is made.
- No messages from any subscriptions are received - so it's not just 1 broken one.
- Services and parameters continue to work
- Refreshing the webpage doesn't do anything
- The only fix is to restart rosbridge server which works immediately.

- Library Version: ros-noetic
- ROS Version: Noetic
- Platform / OS: Ubuntu 20

Thanks for the help, I'm not sure how to go about debugging this one.

remod, 2023-07-14:
Can you reliably reproduce this? Did you find a solution? We are sporadically experiencing the same issue.

lkeselman, 2023-07-14:
> Can you reliably reproduce this? Did you find a solution? We are sporadically experiencing the same issue.

Unfortunately not reliably. I did try to create a minimal scenario and stress test it but could not reproduce it after a couple weeks of continuous running.

My only solution has been to add some automation that allowed the JavaScript client to restart rosbridge when it notices the issue.

jtbandes, 2023-07-14:
Hi folks, not sure if it is relevant to you, but you might be interested in checking out [`foxglove_bridge`](https://github.com/foxglove/ros-foxglove-bridge) which is more actively maintained. There is a [JS/TS client library](https://github.com/foxglove/ws-protocol/tree/main/typescript/ws-protocol) you can use to interact with it from a webpage.

remod, 2023-07-20:
Thank you! I've had a first success with setting `queue_length` to `1` instead of `0` which is the default when a topic is created it [`roslibjs`](https://github.com/RobotWebTools/roslibjs/blob/develop/src/core/Topic.js). This helped me to solve an "entangled connection status" between two clients: When one client disconnected (e.g. disable WiFi), the other client stopped receiving messages as well. Not exactly the issue here, but related.

closed 2024-08-19 by github-actions[bot]