RobotWebTools/rosbridge_suite #831: Rosbridge server hangs. Very very similar to stale issue, and reproducible by the same steps

stribor14, 2022-12-21:
**Description**

- Library Version: 0.11.16-1focal.20221124.044710
- ROS Version: noetic
- Platform / OS: Ubuntu 20.04

**Steps To Reproduce**

As previously discussed in stale (but very detailed issue #425):
- first client on Device 1 through Network 1
- second client on Device 2 through Network 1
- we connect Device 2 to Network 2
- refresh Device 2
- All devices hang (rosbridge server doesn't serve topic msgs to clients anymore)
- interestingly, services still work

We added "queue_length=1" to Subscribers on the client side and it didn't help

**Expected Behavior**
rosbridge_server that works

**Actual Behavior**
rosbridge_server hangs, i.e., topics don't go through. Services still work. Only solution we found is to restart rosbridge_server

stribor14, 2022-12-21:
After additional testing, a single device is enough to reproduce the error:

- run rosbridge_server [nr of clients: 0]
- connect client through Network 1 [nr of clients: 1]
- switch client WiFi to Network 2 [nr of clients: 1]
- reload page (roslibjs) and it will hang [nr of clients: 2]

Normal page reload first removes the client ([nr of clients: 0]), and then creates a new one ([nr of clients: 1]). But when connecting to another network, it doesn't remove the old one, and just creates a new one ([nr of clients: 2]).

The reason for switching networks is that one has internet but is limited in connecting to the robot. The other is the robot hosted network.

stribor14, 2022-12-21:
Furthermore. This doesn't happen if we close the client before changing the networks (since it cleans up connected clients). But in the case when it hangs. We found a new solution. Reconnect to the previous Network 1, open + close the client, and then return to Network 2 and reload the client.

It looks like that after returning to the Network, it detects duplicate from same source and cleans it. When all "stale" clients are cleaned from "connected_clients", rosbridge_server works normally

stribor14, 2022-12-21:
Further minimized steps to reproduce:

- on the client side, set static IP
- run the client [nr of clients: 1]
- change static IP and reconnect
- refresh client, rosbridge hangs [nr of clients: 2]

To fix it, as before, clean the connected clients so that there are no "stale" ones. E.g., change static IP to 1st one, refresh, close client, change static IP to 2nd one, run the client, wait few seconds for rosbridge server to "clean up" connected clients

MiroslawRajczyk, 2023-09-10:
Hi, I was fighting with this issue myself for some time now and here is the reason for this issue and the solution:
This is occurring due to **WebSocketServerProtocol** hanging in some sort of multicasting state as it tries to send the subscribed topics data to all connected clients. As the client's state changed (user changed IP, 1 user disconnected from the network, etc) it is no longer reachable from the client_manager's client list.

It is possible to configure this server such that it pings all connected clients in some interval and drops connections that didn't respond in specified time. To do this all you need to do is to modify rosbridge_websocket,launch and modify 2 params:

`  <arg name="websocket_ping_interval" default="0" /> `
 `  <arg name="websocket_ping_timeout" default="30" />  `

to values that suites your needs, for example:

`  <arg name="websocket_ping_interval" default="1" /> `
 `  <arg name="websocket_ping_timeout" default="4" />  `

Or pass those params from other launchfile. 
In short - the main issue is that **websocket_ping_interval** with a value of 0 causes the server to not verify if the clients are still connected and not dropping stale connections.
Hope it helps anyone looking for a solution to this.

stribor14, 2023-11-27:
Since issue is stale and comment from @MiroslawRajczyk solves it with parameter approach, I'll close this

closed 2023-11-27 by stribor14