RobotWebTools/rosbridge_suite #856: nav_msgs/OccupencyGrid is breaking in Humble

AmmarAlbakri, 2023-04-24:
This used to work without a hitch on ROS 1, the same functionality and the same CODE are not running on ROS 2.

When I try to subscribe to a `/map` topic either using OccupencyGrid or a simple Topic class I get the error:
`[rosbridge_websocket-1] [ERROR] [1682343826.795809774] [rosbridge_websocket]: Exception calling subscribe callback: a bytes-like object is required, not 'str'`

I tried other message types and can confirm they are working, the problem is in the data packets of the map.

- Library Version: latest
- ROS Version: Humble
- Platform / OS: Ubuntu 22.04

**Steps To Reproduce**
1. Publish a map using map_server.
2. Run rosbridge_websocket.
3. Start a JS app that listens to the created map topic.
4. Look for an error in the rosbridge_websocket terminal.

**Expected Behavior**
The `map` topic should be subscribed successfully and data transmitted.

**Actual Behavior**
`[rosbridge_websocket-1] [ERROR] [1682343826.795809774] [rosbridge_websocket]: Exception calling subscribe callback: a bytes-like object is required, not 'str'`

I can provide extra information regarding this problem, Please fix it quickly...

daisukes, 2023-06-01:
I got the same error and found that the ros2 branch needs to include a part of #460.

AmmarAlbakri, 2023-06-01:
@daisukes I can't wait till they merge the PR, Where can I get the fixed repo?

daisukes, 2023-06-01:
@AmmarAlbakri 

You can find it here!

https://github.com/CMU-cabot/rosbridge_suite/tree/ros2-png-compression-fix

AmmarAlbakri, 2023-06-05:
@daisukes I am getting a different error from the package you provided:
`[rosbridge_websocket-1] [ERROR] [1685954630.480407510] [rosbridge_websocket]: [Client 757febfe-afc9-44b8-a60b-5a316aa3396a] [id: subscribe:/map:2] subscribe: Unable to import msg class OccupencyGrid from package nav_msgs. Caused by module 'nav_msgs.msg' has no attribute 'OccupencyGrid'
`

I looked through the code but didn't find the error source, maybe you could help?

daisukes, 2023-06-05:
@AmmarAlbakri 
It looks like your environment does not have the `nav_msgs` package.

AmmarAlbakri, 2023-06-05:
[image]
I do on my setup, do i have to do anything else?

AlfonGio, 2023-06-28:
Hi @daisukes, @AmmarAlbakri,  I have the same issues and follow your solution on the github link you shared. But when I try to run it the same issue still showing `[rosbridge_websocket-1] [ERROR] [1687921966.417605806] [rosbridge_websocket]: Exception calling subscribe callback: a bytes-like object is required, not 'str'`

Launch script `ros2 launch rosbridge_server rosbridge_websocket_launch.xml`
JS app I use is the example app map.html from here: https://github.com/RobotWebTools/ros2djs/blob/develop/examples/map.html

Do I wrongly launch the rosbridge_server web or are there any other missing step that I make? 

Thank you

closed 2023-10-26 by sea-bass

electronicofive, 2025-06-12:
Hi! I'd be happy to help with this. After analyzing PR #861 and testing locally, I found that we can achieve the same JSON serialization fix with a **minimal change**.

**The Issue:** 
- `standard_b64encode()` returns `bytes` object  
- `bytes` is not JSON serializable → causes the map request error

**My Proposed Solution:**
Instead of the full rewrite in #861, we only need **one line change**:

```python
# In encode() function:
return encoded.decode()  # Convert bytes to string
```

in this line:
https://github.com/RobotWebTools/rosbridge_suite/blob/f3baaf24dabbc246d3c83c089c99c195d0a32281/rosbridge_library/src/rosbridge_library/internal/pngcompression.py#L52

**Test Updates:**
- Fixed inconsistency between `str(bytes())` and `str(bytearray())` in [[this line](https://github.com/RobotWebTools/rosbridge_suite/blob/f3baaf24dabbc246d3c83c089c99c195d0a32281/rosbridge_library/test/internal/test_compression.py#L16)](https://github.com/RobotWebTools/rosbridge_suite/blob/f3baaf24dabbc246d3c83c089c99c195d0a32281/rosbridge_library/test/internal/test_compression.py#L16)
- Added `assertIsInstance(encoded, str)` to prevent future regressions after [[this line](https://github.com/RobotWebTools/rosbridge_suite/blob/f3baaf24dabbc246d3c83c089c99c195d0a32281/rosbridge_library/test/internal/test_compression.py#L12)](https://github.com/RobotWebTools/rosbridge_suite/blob/f3baaf24dabbc246d3c83c089c99c195d0a32281/rosbridge_library/test/internal/test_compression.py#L12)

**Testing Results:**
- All 144 existing tests pass: `0 errors, 0 failures`  
- Robot map requests now work (confirmed locally)

**Multi-Branch Question:**
Should I target `ros2` branch first then backport to `humble`, or handle both simultaneously?

I can have the PR ready within a few hours once I know the preferred workflow.

electronicofive, 2025-06-12:
@sea-bass , its done, for ros-2 and humble