RobotWebTools/rosbridge_suite #860: Why was authentication removed in the ROS 2 version of the protocol?

augustelalande, 2023-05-29:
As the title says, why was authentication removed in the ROS 2 version of the protocol? This seems essential for operating in any sort of production environment.

reinzor, 2023-10-24:
@amacneil @jtbandes could you provide some information about this removal? We are in the migration to ROS2 and we relied upon the authentication mechanism that was present in `ros1`. What would be the proper way for us to continue if our app relies on authentication? 

We could bump https://github.com/GT-RAIL/rosauth to `ros2` if desired and re-implement the auth mechanism in this repo. What are your thoughts?

amacneil, 2023-11-16:
From memory it was broken, and we wanted to focus on getting a ROS 2 version working, especially for use cases like Foxglove where 99% of the time you are connecting to localhost or a local IP address and authentication is not a concern.

We have not been maintaining this library lately, since we had unresolvable performance issues with rosbridge we created a separate [ROS Foxglove Bridge](https://docs.foxglove.dev/docs/connecting-to-data/ros-foxglove-bridge/). So feel free to make whatever PR proposals you like to the project and the current maintainers can take a look.

reinzor, 2023-11-16:
@amacneil ; thanks for the update / background.

closed 2023-11-16 by augustelalande