RobotWebTools/rosbridge_suite #885: OccupancyGrid' object has no attribute '_slot_types' #638

jesusramondovale, 2023-10-23:
- Library Version: latest 
- ROS Version: Humble
- Platform / OS: Ubuntu 22.04

I'm using ros3djs on my ASP.NET webpage (cshtml) and when I try to visualize a 3D map the ros_server service executed in my ROS machine stops and gives the error : 
```py
"OccupancyGrid' object has no attribute '_slot_types'"

`File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosbridge_library/capabilities/subscribe.py", line 159, in _publish
    self.publish(message, self.fragment_size, self.compression)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosbridge_library/capabilities/subscribe.py", line 318, in publish
    outgoing_msg = message.get_cbor(outgoing_msg)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosbridge_library/internal/outgoing_message.py", line 38, in get_cbor
    outgoing_msg["msg"] = self.get_cbor_values()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosbridge_library/internal/outgoing_message.py", line 33, in get_cbor_values
    self._cbor_values = extract_cbor_values(self._message)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosbridge_library/internal/cbor_conversion.py", line 53, in extract_cbor_values
    for slot, slot_type in zip(msg.__slots__, msg._slot_types):
AttributeError: 'OccupancyGrid' object has no attribute '_slot_types'`
```

dzw-tw, 2024-04-11:
I also have this problem, have you solved it?

chanhhoang99, 2024-04-23:
For whom who get this error, I believe you are installing your rosbridge_suite from apt which is out of date.
I tried building rosbridge_suite from source, seems to work fine at least with ros3djs.
Of it's still not working, you may try to set compression value to false

jesusramondovale, 2024-05-29:
> I also have this problem, have you solved it?

It was a .js file versions conflict

luojinzhang, 2024-06-24:
I am having the same thing, I indeed installed it using `sudo apt-get install ros-humble-rosbridge-suite`, any official fixing on this ? For workaround I guess it would be building the latest rosbridge_suite locally ? Thank you.

jesusramondovale, 2024-06-25:
My problem was a conflict between versions on client side (JavaScript imported files from ros3djs) not the ROS bridge.

jesusramondovale, 2024-06-25:
> I am having the same thing, I indeed installed it using `sudo apt-get install ros-humble-rosbridge-suite`, any official fixing on this ? For workaround I guess it would be building the latest rosbridge_suite locally ? Thank you.

My case isn't a problem of ROS-Bridge, but you could try increasing the parameter "maximum fragment size" on ros-bridge configuration (or something like that) 

When a map is bigger than the fragment size, it needs more than one fragment and the Ros JS client side doesn't work well with that. I think you have to manually rebuild the data of multiple fragments into one on client side, or basically increase the fragment size on server side in order to avoid this

closed 2025-07-26 by github-actions[bot]