RobotWebTools/rosbridge_suite #888: TypeError: 'type' object is not subscriptable when launching rosbridge_websocket

AlfonGio, 2023-11-14:
**Description**
Hi, I am trying to launch rosbridge server: rosbridge_websocket to connect ROS2 environment with web interface but this error arise.

- Library Version: ros2 version rosbridge_suite repo
- ROS Version: Foxy
- Platform / OS: Jetson Orin Nano / NVIDIA Jetpack 5 / Ubuntu 20.04

**Steps To Reproduce**

1. Clone ros2 branch of rosbridge_suite locally
2. Colcon build the rosbridge_suite package
3. Source the workspace
4. Launch the package using `ros2 launch rosbridge_server rosbridge_websocket_launch.xml`

**Expected Behavior**
The rosbridge_server successfully launched

**Actual Behavior**
[rosapi_node-2] Traceback (most recent call last):
[rosapi_node-2]   File "/home/orin/workspaces/web_ros-dev/install/rosapi/lib/rosapi/rosapi_node", line 41, in <module>
[rosapi_node-2]     from rosapi import glob_helper, objectutils, params, proxy
[rosapi_node-2]   File "/home/orin/workspaces/web_ros-dev/install/rosapi/lib/python3.8/site-packages/rosapi/objectutils.py", line 39, in <module>
[rosapi_node-2]     from rosbridge_library.internal import ros_loader
[rosapi_node-2]   File "/home/orin/workspaces/web_ros-dev/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/ros_loader.py", line 132, in <module>
[rosapi_node-2]     typestring: str, intf_type: str, loaded_intfs: dict[str, Any], intf_lock: Lock
[rosapi_node-2] TypeError: 'type' object is not subscriptable

[image]

sea-bass, 2023-11-14:
I suspect that because you're on Foxy (an unsupported ROS 2 version), that uses Python 3.8, type hints are being affected as shown in e.g. this issue:

https://github.com/langchain-ai/langchain/issues/11226#issuecomment-1744288441

AlfonGio, 2023-11-15:
It should works, the websocket_launch.py can be launched. Thanks.

[image]

closed 2023-11-15 by AlfonGio