RobotWebTools/rosbridge_suite #898: Cut a new release?

sea-bass, 2023-12-20:
With the ROS 2 test porting (and according fixes) and action support we recently added, we should probably cut a new release in this repo as well.

This could be 1.4.0, but given we're doing a v2 in `roslibjs`, maybe this could be 2.0.0 as well?

lukeopteran, 2024-01-17:
@sea-bass a release would be great, we would like to use the working image compression, without having to clone to a specific commit.

The latest ros2 branch produces a type error in the ros_loader (Running under Galactic!). It works fine at commit 501a926

[image]

Thanks

danmartzla, 2024-06-28:
Hi all,

Is the new release planned to be launched soon? I need to use actions through "rosbridge" but the latest release available on the debian repositories don't support them yet.

Thank you.

sea-bass, 2024-06-28:
You can always build this repository from source and use the actions support today.

sea-bass, 2024-08-16:
For a bit of background, this project's ROS 2 action support was driven by needing them in `roslibjs`.

The working plan is that we need some help writing integration tests for ROS 2 in `roslibjs` (https://github.com/RobotWebTools/roslibjs/issues/746), and once that is clear, we were going to release a version 2 of both these repos.

scastro-bdai, 2024-09-27:
As mentioned in my previous comment, it was decided that some ROS 2 tests for `roslibjs` are needed before officially cutting the v2 release of this tool... but the way things are shaking out, it's unclear to me if anyone will actually contribute those changes, so maybe worth just releasing `rosbridge_suite` for now?

Tagging @EzraBrooks @MatthijsBurgh

sea-bass, 2024-09-27:
As with my last comment, the plan was to make some ROS 2 tests for `roslibjs` and then cutting a release for both that and `rosbridge_suite`.

But the way things are shaking out, seems nobody is willing to contribute those tests, so I'd be open to maybe just releasing `rosbridge_suite` at this point.

@EzraBrooks @MatthijsBurgh what do you guys think?

EzraBrooks, 2024-09-27:
I don't think blocking `rosbridge_suite`'s release on `roslibjs` makes sense, since other roslib implementations use it. I'm still not interested in releasing `roslibjs` until it has ROS 2 integration tests, but I don't think that should gate this repo.

sea-bass, 2024-10-03:
@bjsowa Now that you're officially approved for releases, would you like to help cut a `2.0.0` release of this repo? I can prepare the changelogs and update version numbers, but I don't have access to actually bloom the repos.

The tricky thing is that the `humble` branch actually has diverged from `ros2` (which includes iron+later), so I'm not sure what the versioning should be.

This is because https://github.com/RobotWebTools/rosbridge_suite/pull/932 was needed for Jazzy, but did not work on Humble, so we did this: https://github.com/RobotWebTools/rosbridge_suite/pull/934

I guess what makes most sense to me is:

* Backport everything that is possible to `humble` branch, tag that as `2.0.0`
   * All future updates to `humble` will necessarily be patch releases, so `2.0.1`, ...
* Tag the `ros2` branch as `2.1.0`

bjsowa, 2024-10-03:
@sea-bass Sure, I can run bloom if you prepare the branches.

> * Backport everything that is possible to `humble` branch, tag that as `2.0.0`
>   
>   * All future updates to `humble` will necessarily be patch releases, so `2.0.1`, ...
>   * Release this to Humble
> * Tag the `ros2` branch as `2.1.0`
>   
>   * Release this to Iron, Jazzy, and Rolling

This is what I typically do in such case, except I increase the major number so `ros2` branch would get `3.0.0` version. This is not ideal though if the major version has some special semantic meaning to you.

sea-bass, 2024-10-03:
Yeah, I might stick with 2.1.0 for the ros2 version since the changes will truly be minor.

sea-bass, 2024-10-03:
OK all the backports needed for humble are [here](https://github.com/RobotWebTools/rosbridge_suite/pulls?q=is%3Apr+is%3Aopen+base%3Ahumble+).

Once these are in, I'll generate a `2.0.0` changelog from `humble`, then we'll bring that up to `ros2` and amend it to `2.1.0` with the few updates.

@MatthijsBurgh -- it seems like @bjsowa was only added to the `web_video_tools` repo, but not to this one as a possible reviewer. Could you look into that as well? Thanks!

MatthijsBurgh, 2024-10-07:
@sea-bass Done. I think we should reorder all the organization teams soon. As I think it is a pretty big mess.

sea-bass, 2024-10-08:
> @sea-bass Done. I think we should reorder all the organization teams soon. As I think it is a pretty big mess.

It doesn't look too bad -- the file is here: https://github.com/ros2-gbp/ros2-gbp-github-org/blob/latest/robotwebtools.tf

closed 2024-10-08 by sea-bass

sea-bass, 2024-10-08:
#966 and #967 have been merged, and the releases will be added to the next syncs of Humble, Iron, Jazzy, and Rolling.

MatthijsBurgh, 2024-10-08:
@sea-bass I meant the teams in the github organization, https://github.com/orgs/RobotWebTools/teams,.