RobotWebTools/rosbridge_suite #903: when send action goal , gets an error and tried to find action interface in the msg packages

Cra2yQi, 2024-01-24:
**Description**

- Library Version: ros2
- ROS Version: humble
- Platform / OS: MacOs / ubuntu22.04

**Steps To Reproduce**

when i use roslibjs to send a action goal ，rosbridge-suit gets an error and tried to find action interface in the msg packages.

  ```javascript
action(serverName, actionName, argument) {
    const rosAction = new ROSLIB.ActionClient({
      ros: this.ros,
      serverName: serverName,
      actionName: actionName
    })
    const goal = new ROSLIB.Goal({
      actionClient: rosAction,
      goalMessage: argument
    });
    // Print out their output into the terminal.
    goal.on('feedback', function (feedback) {
      console.log('Feedback: ' + feedback.sequence);
    });
    goal.on('result', function (result) {
      console.log('Final Result: ' + result.sequence);
    });

    goal.send()
  }
```
[image]

**And I tried to use python to send the action, and I got the same error**

```python
import roslibpy
import roslibpy.actionlib
import time

def feedback_callback(feedback):
    print('Feedback received: ', feedback['sequence'])

def result_callback(result):
    print('Result received: ', result['sequence'])

def call_fibonacci_action(order):
    client = roslibpy.Ros(host='10.211.55.5', port=9090)
    client.run()

    action_client = roslibpy.actionlib.ActionClient(client, '/fibonacci', 'actionlib_tutorials/Fibonacci')

    goal = roslibpy.actionlib.Goal(action_client, roslibpy.Message({'order': order}))

    goal.on('feedback', feedback_callback)
    goal.on('result', result_callback)

    print('Sending goal...')
    goal.send()

    # 等待结果
    while not goal.is_finished:
        time.sleep(1)

    print('Goal processing done.')
    client.terminate()

if __name__ == '__main__':
    call_fibonacci_action(5)
```
[image]

sea-bass, 2024-01-24:
ROS 2 action support was added recently to `rosbridge_suite` in https://github.com/RobotWebTools/rosbridge_suite/pull/886. and is only implemented in `roslibjs`, not `roslibpy` yet.

The APIs you are using are the ones that work in ROS 1, where indeed the `.msg` namespace is searched to publish to the action's hidden topics.

sea-bass, 2024-01-24:
I've also created the underlying issue for `roslibpy` here: https://github.com/gramaziokohler/roslibpy/issues/122

So will go ahead and close this one, if that's okay.

closed 2024-01-24 by sea-bass