RobotWebTools/rosbridge_suite #910: 'float' object has no attribute 'get_fields_and_field_types'

Hytac, 2024-02-28:
**Description**
When calling the `/get_planning_scene` service (moveit_msgs/srv/GetPlanningScene) it fails to serialize the results with the followign error:

`[rosbridge_websocket-1] [ERROR 1709106045.616655214] [rosbridge_websocket]: [Client 8b631744-1ef3-4af3-8555-3e8281b42001] [id: call_service:/get_planning_scene:8] call_service AttributeError: 'float' object has no attribute 'get_fields_and_field_types' (() at :0)`

Just to mention this PR fix the issue: [#840](https://github.com/RobotWebTools/rosbridge_suite/pull/840).

- Library Version: Build from source (7d78af16d30d0ffe232abcc65d0928ce90bd61f7)
- ROS Version: Humble
- Platform / OS: 22.04

**Steps To Reproduce**
I've seen others to have this error with different messages, but using moveit it is easy to make it fail, Just call the service get_planning_scene

tilt-silvie, 2024-03-08:
I have same issue.

tilt-silvie, 2024-03-09:
@Hytac I made a PR for this issue. I haven't verified the same steps that you've written, but I hope this PR fixes your problem!

EzraBrooks, 2024-06-07:
Hi, I've just reopened #840 for this fix - I erroneously thought that the issue had been fixed, because of an unrelated code change in my application that made it stop happening. I'll try to get 840 in ASAP to fix for good.

PR #840 (2024-08-15): Fix length-limited list types
**Public API Changes**

None

**Description**

Removes the `, N` length specifier at the end of list types like `sequence<type, length>`. Fixes issues with serializing `shape_msgs/Polygon`, which has a `float64[<=3]` field.

Fixes #910
closed 2024-08-15 by EzraBrooks