RobotWebTools/rosbridge_suite #917: Use pydantic instead of basic_type_check

yoelbassin, 2024-04-05:
Currently the `rosbridge_library` doesn't have any type hints (PEP 484). Type hints enhance the readability and maintainability of the codebase. 

In the `Capability` class in `capability.py`, which all the capabilities inherit from, there is a `basic_type_check` function which performs type checking for the message fields the capability receives. 

Instead of using this function, we can use `pydantic` which will 1) perform the type checking and 2) add type annotations for the message fields. 

I think this could improve the codebase quality and would remove redundant code.

I would be happy to work in this PR :)

yoelbassin, 2024-04-05:
@sea-bass would love to hear your opinion about this one. BTW are there any other active maintainers?

EzraBrooks, 2024-04-05:
I'd be happy to help review a PR for this. The serialization and deserialization in this package is pretty hard for me to wrap my head around - especially where there's special cases for certain types, which exist in a few places - and it would be great to have static type checking to help trace through it.

yoelbassin, 2024-04-05:
> especially where there's special cases for certain types, which exist in a few places

Could you mention some of the places you've seen this, so I would skim through them? :)

yoelbassin, 2024-04-06:
Well, after playing with it a little I found the `pydantic` version that `rosdep` supports is at most [`1.10`](https://pkgs.org/search/?q=python3-pydantic) which is a little outdated. The newest `pip` version is `2.6` where breaking changes were introduced since the `1.*` versions.

I'm not sure what approach should I take, either implementing a simpler version of `pydantic` on `dataclass`, or using the older `pydantic` version.

What do you think would be the better approach?

EzraBrooks, 2024-04-08:
here's one such odd place, where the library implicitly converts uint8[] to a base64 string. This one in particular actually bit me recently on a project.

https://github.com/RobotWebTools/rosbridge_suite/blob/7d78af16d30d0ffe232abcc65d0928ce90bd61f7/rosbridge_library/src/rosbridge_library/internal/message_conversion.py#L203

EzraBrooks, 2024-04-08:
It looks like [Pydantic is at version 1.10 even on Ubuntu Noble](https://packages.ubuntu.com/noble/python3-pydantic), which releases this month (April 2024), so it seems we should target the older version.

closed 2025-05-09 by github-actions[bot]